micropython/micropython #6396: ESP32: Bluetooth together with machine.lightsleep()

mbuesch, 2020-09-02:
Is Bluetooth supposed to work, if the main loop of the program uses ``machine.lightsleep(milliseconds)``?

My test program runs a simple gatt peripheral and the main loop is empty (for testing) and just sleeps for 500 milliseconds.
This works, as long as I use ``time.sleep(0.5)``. BT connection works just fine with that.

If I change the sleep to ``machine.lightsleep(500)`` it stops to work correctly.
I'm unable to connect to the BT after changing to lightsleep.

Are the bluetooth interrupts unable to wake the system (in time) for a correctly functioning BT?

How else could I conserve power, if lightsleep does not work?

dpgeorge, 2020-09-03:
This requires some investigation, to see how to reduce power consumption with BT enabled.  As a starting point I'd suggest looking if the IDF provides examples (in C) of how to do low-power BLE, and then using that as a guide how to do it in MicroPython.

tve, 2020-09-03:
> How else could I conserve power, if lightsleep does not work?

I know nothing about BT on esp32, but the wifi gets cut during lightsleep. The solution there is to use automatic light sleep, which lets the processor wake up periodically to stay connected to the AP. I wonder whether for BT it's similar? There's a PR pending to enable automatic light sleep and the comments discuss some of the issues. (Hope this helps.)

Carglglz, 2020-09-04:
@tve
> I wonder whether for BT it's similar? 

Yes for BLE is similar, one can select (at connection time) the *Connection interval* at which data is transmitted (defined by the master device *). I think it is in the range of 7.5 ms - 4s and it is managed by the Link Layer. 

There is also *Slave latency* and *Connection supervision timeout* , (see [Connection parameters](https://devzone.nordicsemi.com/f/nordic-q-a/28058/what-is-connection-parameters))

*A change may be requested by the peripheral after that. 
(*L2CAP_CONNECTION_PARAMETER_UPDATE_REQ* in Bluetooth 5.2 specification, Volume 3, Part A, Section 4.20)

@dpgeorge 
Here are the examples:
[esp-idf connection parameters update](https://github.com/espressif/esp-idf/issues/672)

@mbuesch 
> How else could I conserve power, if lightsleep does not work?

Setting this along with the *Transmit Power* would allow the lowest operation mode possible. 👍

xyzzy42, 2024-02-20:
This can be done with automatic light sleep.  It does dramatically reduce power consumption, to about 2 mA to maintain a BLE connection or to have periodic advertising on.  Of course, if one transmits a lot it uses more power.  I've written the code to do this in Micropython and it works, with a big caveate. 

µpy asyncio uses busy-polling.  So if you use aioble, and thus asyncio, µpy will run continuously without ever sleeping, even if all asyncio tasks are sleeping and waiting for input or timers.  And power consumption is thus much higher since light sleep can not be used.  It's no lower than it it without automatic lightsleep enabled.

smhodge42, 2024-02-20:
This is somewhat off-topic, but what do you mean by automatic light sleep – with the emphasis on “automatic”? I thought the code had to explicitly put it into light sleep each time.

xyzzy42, 2024-02-20:
> This is somewhat off-topic, but what do you mean by automatic light sleep – with the emphasis on “automatic”? I thought the code had to explicitly put it into light sleep each time.

The ESP-IDF has a feature to automatically enter lightsleep when the CPU (really FreeRTOS) is idle for more than a couple ticks.  Then automatically resume when something happens and keep going from where it left off.

You don't need to ever manually enter lightsleep.  One just magically draws 2 mA when waiting for BLE connection, waiting for GPIOs, waiting for GAP operations on a BLE connection and so on.

But since asyncio doesn't work with it, it's practically useless on µpy.

smhodge42, 2024-02-20:
What if you are not using asyncio?