micropython/micropython #6397: Roadmap to next release, v1.14

dpgeorge, 2020-09-03:
This is an issue to discuss and plan for the next release.  The idea is to aim for a 2 month release cycle, so the next release would be the first week of November 2020.  See previous release ticket for v1.13 at #6271 

To keep focus and set expectations, a release cycle would have a main goal (eg a set of related new features) which would be the priority.  But bug fixes and other items can still be considered.

I know there will be a lot of competing requests for features.  There are also a lot of outstanding items.  But progress needs to be made and for the next release I propose to focus on:
- efficient polling with "uevent", and external events in `uasyncio`
- non-blocking SSL support in `uasyncio`

Secondary items to focus on would be:
- definition and implementation of a common `machine.PWM` class
- moving to an Epoch of 1970

Other items still can, and still will, be worked on.  But in the interest of having more strict release deadlines there needs to be some primary goals to make the releases meaningful.

tve, 2020-09-03:
FYI, while I like the topics (having worked on 3 of the 4) I will need some scheduling in order to help. I'm pretty busy with projects using MP :-) but have flexibility. What kills me, to be honest, is when it takes a week to get feedback and so I have to reassemble everything from test setup to my memory to make the next round of changes. So if there's a way to coordinate and focus for a couple of days on one issue it would make it a lot easier for me to contribute (which you're free to ignore too).

dpgeorge, 2020-09-04:
> So if there's a way to coordinate and focus for a couple of days on one issue it would make it a lot easier for me to contribute (which you're free to ignore too).

I understand.  Different timezones and remote working can make it hard.

uevent is tough, so I think we should focus on non-blocking SSL first.  I will review your PRs on that.

beyonlo, 2020-09-04:
@dpgeorge and @tve 

> * non-blocking SSL support in `uasyncio`

I'm working in the application with socket (non-blocking) using uasyncio v3,  following exactly the @peterhinch very good examples:
Client: https://github.com/peterhinch/micropython-async/blob/master/v3/as_drivers/client_server/uclient.py
Server: https://github.com/peterhinch/micropython-async/blob/master/v3/as_drivers/client_server/userver.py

That works fine to me, but I would like SSL because I use them over Internet. I'm using ESP32.

Well, that examples above use:
```
sreader = asyncio.StreamReader(sock)
swriter = asyncio.StreamWriter(sock, {})
```

Question: with this feature new feature ( non-blocking SSL support in `uasyncio`), these examples above will works at same way, using asyncio.StreamReader, with SSL?

Thank you so much.

dpgeorge, 2020-09-17:
>  with this feature new feature ( non-blocking SSL support in `uasyncio`), these examples above will works at same way, using asyncio.StreamReader, with SSL?

Yes, it should work transparently, the socket will act exactly the same whether it is SSL or non-SSL (or any other kind of stream, eg UART).

t35tB0t, 2020-10-29:
Late entries (been away), possibly push to 1.15?
* Add signal.py cPython still signals (interrupts) support:  keyboard SIGQUIT   (CTRL-\ ) and maybe SIGTRAP as minimums
* Add better UDP datagram support to asyncio

Dr-StephenStrange, 2020-11-06:
Would it be possible to get webrepl properly working on the Pyboard D?

At the moment nobody can connect it the main thread is running

street-grease-coder, 2020-11-18:
hope it gets released soon, waiting eagerly, thanks for the hard work

dpgeorge, 2021-02-02:
v1.14 was tagged in 78b23c3a1f064dc19bbee68930ef3aba110c781c on 2nd Feb 2021... a bit later than planned... next release I'll try to be more strict about it.

Above feature requests have been noted.

closed 2021-02-02 by dpgeorge

beyonlo, 2021-02-08:
@dpgeorge Thanks for the works.

Just a clarification: 

I see in the announce 1.14 version ( https://github.com/micropython/micropython/releases/tag/v1.14 ) that PPP is now working in IDF >= 4, for ESp32 port:
"modnetwork: re-enable PPP support for IDF-SDK >=v4"

But in the download area ( http://micropython.org/download/esp32/) show that is not working yet:
"
Firmware with ESP-IDF v4.x
Firmware built with ESP-IDF v4.x, with support for BLE, but no LAN or PPP:
"

Please, could you clarify that?

Thank you so much to you and all MicroPython developers for the amazing work.