micropython/micropython #6409: I can't use this library on my Esp8266 board

sharifpt, 2020-09-05:
I can't use this library on my Esp8266 board. I am run sdtest.py file.  But it’s not respons,  not do anythins.

robert-hh, 2020-09-05:
Which library are you talking about? Did you adapt the file Drivers/sdtest.py to the ESP8266, especially line 9?
Which ESP8266 board are you using and which SD card adapter?

Besides that, questions like these are better asked in the forum: https://forum.micropython.org/

sharifpt, 2020-09-05:
sdcard.py library i can't run esp8266.  I can't sd interfacing with esp8266.  Please help me.

robert-hh, 2020-09-05:
Yes, we understand. 
First question: Which ESP8266 board are you using. Which SD card adapter?

robert-hh, 2020-09-05:
The file sdtest.py from the repository does not work with an ESP8266. Line 9 has to be changed from 
```
    sd = sdcard.SDCard(spi, machine.Pin.board.X21)  # Compatible with PCB
```
to 
```
    sd = sdcard.SDCard(spi, machine.Pin(15)) # esp8266
```
Assuming that CS of the CS card is connected to GPIO15, which is for instance D8 on a Wemos D1 Mini. CS may be connected to other pins.
The SD card has to be FAT formatted with exactly one partition.

sharifpt, 2020-09-06:
[image]
I am run this code use thonny python and micropython v1.13 on 2020-09-02; 
But this code do nothing. 
Please give me a code. Then i can brower my sd card folder only..  
Thamks

robert-hh, 2020-09-06:
You have to import the sdtest modules and then call the sdtest() method. Something like the following two lines, which you can call  in a main file for enter at the REPL prompt.
```
import sdtest
sdtest.sdtest()
```

sharifpt, 2020-09-06:
@robert-hh 
[image]
Give me some error.... Please help me....

peterhinch, 2020-09-06:
The ENODEV error means that the program can't detect your SD card. You need to adapt **sdtest.py** to match the wiring between the ESP and the SD card.

As @robert-hh  has pointed out, this support request should be in [the forum](http://forum.micropython.org/index.php) rather than this bug tracker.

robert-hh, 2020-09-06:
I repeat the same question again the third time, this time loud:
WHICH ESP9266 MODULE ARE YOU USING, AND HOW IS THE SD CARD ADAPTER CONNECTED TO THE ESP8266? If you do not know the name, make a picture of the top side and the bottom side.
Without that information it is hard to help you.

sharifpt, 2020-09-06:
[image]
I am useing NODEMCU V3 and this type sd card module
[image]
 
[image]

My modules ok. I'm test this module arduino uno   
[image]

My pin connection is
esp8266  	 sd card Module 
D5   	  	clk
D6  	  	 	MISO	
D7   	  	MOSI
D8  	  	  	cs

3v   	  	  3v3
gnd  	 gnd

robert-hh, 2020-09-06:
The wiring looks ok. You could cross check again the cables and the bread board. The error still looks like the card is not detected at all. I cannot tell whether the SD card is fully pushed into the socket. And, as said, the SD card must have only one FAT partition. 

Besides that not all SD cards work. I used successful 8GB and 16GB cards, but others failed

sharifpt, 2020-09-06:
@robert-hh  Thanks for helping and support.  You'll try to give me a solution .     Maybe my esp8266 board problem or other.  Few days l am buy new esp board and sd card module. Then I'll try your    code and flowing your instruction.  I am a student and beginner with micropython. If i'll  to hurt you,  please forgive me. Thank you again.

closed 2020-09-06 by sharifpt