micropython/micropython #6414: nrf52 DK Bluetooth Support in Micropython (Zephyr Port)

mridwang, 2020-09-06:
Hi everyone, I am getting the following error while importing Bluetooth Module in NRF52 DK in REPL. Pleas advise. 
>>> import bluetooth
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'bluetooth'

robert-hh, 2020-09-06:
Did you look into the nrf examples at https://github.com/micropython/micropython/tree/master/ports/nrf/examples? They import a module with the name "ubluepy".

Besides that: questions like these are better asked in the forum at https://forum.micropython.org/
This place here is the bug tracker.

mridwang, 2020-09-06:
> Did you look into the nrf examples at https://github.com/micropython/micropython/tree/master/ports/nrf/examples? They import a module with the name "ubluepy".
> 
> Besides that: questions like these are better asked in the forum at https://forum.micropython.org/
> This place here is the bug tracker.

Thanks. I have tried to import "ubluepy" getting the same error.

MaureenHelm, 2020-09-11:
The Bluetooth module isn't supported in the Zephyr port yet.

jonnor, 2024-07-13:
Since April 2021 there is a bluetooth module for Zephyr. So I believe this issue can be closed.

closed 2024-12-27 by jonnor