micropython/micropython #6415: extmod/uasyncio Interface to ISR is problematic

peterhinch, 2020-09-06:
This arose from [this forum post](https://forum.micropython.org/viewtopic.php?f=2&t=9000&p=50766#p50744). The following code sample only runs the task scheduled by the ISR when the one running task wakes up. Tested on a Pyboard 1.1 running firmware V1.13.
```python
import uasyncio
from machine import Pin
import micropython

async def example_task():
    print('example task running')

def task_launcher(_):
    print('task_launcher running')
    uasyncio.create_task(example_task())

def irq_handler(pin):
    print('IRQ!', pin.value())
    micropython.schedule(task_launcher, None)

trigger = Pin(Pin.board.X6, Pin.IN, Pin.PULL_UP)
trigger.irq(irq_handler, Pin.IRQ_FALLING | Pin.IRQ_RISING)

async def endless_loop():
    while True:
        print('Tick.')
        await uasyncio.sleep(5)
        print('Tock.')
        await uasyncio.sleep(5)

uasyncio.run(endless_loop())
```
Given that https://github.com/micropython/micropython/pull/6106 is not merged, the problem can't be fixed with an **Event**.

Is there a way to interface **uasyncio** with an ISR other than a busy-wait loop?

dpgeorge, 2020-09-08:
This is a nice example/use-case, of interfacing IRQs with a cooperative scheduler (ie uasyncio).  But it has the same problems as other attempts to do this, namely that uasyncio doesn't know that the IRQ made something runnable, and also there are possible race conditions (adding the new task may occur in the middle of adding another task).

In CPython I think you want to use `loop.call_soon_threadsafe(lambda: loop.create_task(example_task()))` but it's exactly the implementation of `call_soon_threadsafe()` that is difficult.

dpgeorge, 2020-09-08:
> Is there a way to interface **uasyncio** with an ISR other than a busy-wait loop?

On bare-metal it's possible to use `uio.IOBase` to make an object which can be added to uasyncio's poller.  That's really the only way to do it at the moment.

peterhinch, 2021-03-11:
Closing as fixed by ThreadSafeFlag.

closed 2021-03-11 by peterhinch

scy, 2021-03-16:
> Closing as fixed by ThreadSafeFlag.

For reference, that’s #6886.