micropython/micropython #6417: Disable the REPL on UART(0) on the esp8266: question

EnzoPython, 2020-09-06:
Hi Damien,
I runned this micropython snippet you coded:
import io
import os
class UARTHalf(io.IOBase): 
    def __init__(self, uart):
        self.uart = uart
    def write(self, data):
        return self.uart.write(data)
def test():
    uart = os.dupterm(None, 1)
    uart_half = UARTHalf(uart)
    os.dupterm(uart_half, 1)
    print('testing')
    while True:
        data = uart.read(10)
        if data:
            print('got', data)
        if data == b'q':
            break
    os.dupterm(uart, 1)
test()
I need to exchange a few bytes between an ESP8266 and another board (now using Arduino for ease of testing) and have troubles with REPL. I'm using Thonny IDE to write and upload to ESP8266.
When I run the code from Thonny I lose connection with it because of the uart 0 disable, so I used Putty and it's fine but can never get out with "q" as I should: it goes on reading bytes sent from Arduino and doesn't get anything from the keyboard into Putty.
Please, can you helpme? I'm not expert about ESP!
Basically I need to code a very simple function sending and getting a few tens bytes via serial. It would be great if I could use another serial on the ESP8266 leaving UART0 for undisturbed REPL debugging. I could use it on the final product, too. However any hint to go on will be very appreciated!
Thanks in advance and congratulations for the huge amount of work you do!

robert-hh, 2020-09-06:
ESP8266 has only on fully functional UART. The second UART can only be used for writing.
About not terminating the loop: Since you have that print statement: what is actually printed when you enter q? Maybe data is longer than 1 byte. You could try to read one character at a time.
If you do not want to interfere with REPL, you can also use sys.stdin.read() and sys.stdout.write() for reading and writing to UART 0. The only drawback, that sys.stdin.read() is blocking. It will not return until data is entered.

Two hints on posting:
- If you include code in two lines with three backticks ```, the formatting is maintained.
- The best place for questions like these is the micropython forum. https://forum.micropython.org

EnzoPython, 2020-09-07:
Thanks Robert!
I'm going nut to code this simple task and losing a lot of time!
I don't know how to use Webrepl: I see it accepts an IP address but my ESP8266 is on a private net (10.0.0.x). How can I tell Webrepl to connect to it?
I'll try sys.stdin.read() with a timeout char sending from the other board to avoid getting stuck if no bytes are sent.

robert-hh, 2020-09-07:
If you PC is on the private net too, you can simpy set the IP address in Webrepl to the value of the ESP8266, like 10.0.0.x:8266, and then push the connect button. On the esp8266, webrepl has to be configured once. During that the code to start it will be added to boot.py. Note however, that while a script is running there is no REPL support either way.

pyforever, 2020-09-07:
I tried both [ws://10.0.0.191:8266/](url) and [http://10.0.0.191:8266/](url) and the button changed to "disconnect" but trying to retrieve a file from the ESP stayed in "getting" forever. Moreover I have no prompt and I can't digit anything in the black space.
Regarding the output from the loop I tried sending "q" in different ways from Arduino: char(113) with and without "\r\n", simply "q" but it get catched from the uart.read and always printed as a "q" but it doesn't break the loop.
The only way I found for now is creating a window of some msec waiting for the other board sending and then exit the loop and restoring the standard REPL.

robert-hh, 2020-09-08:
You have only a REPL prompt in webrepl if you would also have a REPL prompt on UART. While you run a script on the ESP8266, there is no feedback in webrepl. I cannot tell what you are doing, but the little script below based on your test script works on my ESP8266 board. And using webrepl I also see the output from the loop's print statement. Beware that the input for the loop has to come from the UART interface, not from webrepl.
```
import os
def test():
    print('testing')
    uart = os.dupterm(None, 1)
    while True:
        data = uart.read(10)
        if data:
            uart.write('got %s\r\n' % data)
            print("got", data)
            if data == b'q':
                break
    os.dupterm(uart, 1)

test()
```

dpgeorge, 2020-09-17:
As mentioned above, these questions are best suited to the forum (here is for bug reports), so I'll close this issue.

BTW, using `sys.stdin.buffer` and `sys.stdout.buffer` are good ways to achieve input/output on the UART.  And you can check for any data on the input using:
```python
import select
rd, _, _, = select.select([sys.stdin.buffer], [], [], 0)
if rd:
    data = sys.stdin.buffer.read(1)
```

closed 2020-09-17 by dpgeorge