micropython/micropython #6419: unix: sys.path affects loading of frozen modules

jimmo, 2020-09-08:
Example:
1. Add a frozen module to unix/windows port.
2. At the repl:

```
import mymodule # works
```

3. create demo.py containing

```
import mymodule
```

Run

```
$ ./micropython ~/mpy/demo.py
Traceback (most recent call last):
  File "/home/jimmo/mpy/demo.py", line 5, in <module>
ImportError: no module named 'mymodule'
```

The underlying issue here is that the unix port, when running normally has `sys.path=["", "~/.micropython/lib:/usr/lib/micropython"]`, and when running with a path to /path/to/demo.py has `sys.path=["/path/to", "~/.micropython/lib:/usr/lib/micropython"]`

When `import mymodule` is executed, it's trying to load `mp_frozen_stat("/path/to/mymodule")`, but the frozen module is just named "mymodule".

I'm not sure if the fix here is to improve the frozen module matching (i.e. assume that all frozen modules are in all sys.paths), or to always have "" as the final entry in sys.path? Or something else? Need to consider how thiss interact with the behavior where filesystem modules override frozen modules of the same name.

This came from the forum: https://forum.micropython.org/viewtopic.php?f=12&t=8995&p=50840#p50840

RobertoRoos, 2020-09-08:
Is there a quick fix to simply append "" to sys.path whenever micropython starts? Is there an accessible way to run start-up code?

(Hitman1O1 from the forum here)

dpgeorge, 2020-09-08:
This is kind of a feature, in that the location of `""` in `sys.path` allows you to control the order of searching in frozen code.  But I think it needs to be improved.

See existing issues: #1804, #2322, #3509

RobertoRoos, 2020-09-08:
For other people who would like a quick and dirty fix, I 'fixed' the feature by modifying the default sys.path value.
In `ports/unix/mponfigport.h` I _added_ the following line:

`#define MICROPY_PY_SYS_PATH_DEFAULT ":~/.micropython/lib:/usr/lib/micropython"`

And in `ports/windows/mpconfigport.h` is _modified_ the line to look like:

`#define MICROPY_PY_SYS_PATH_DEFAULT ";~/.micropython/lib"`

Now `""` is always in sys.path. It's actually in there twice when starting the REPL, but I haven't found a consequence of this yet.

chrisdecker1201, 2021-11-26:
I had/have the same issue.

Quick workaround for me was to just append "" to the sys.path

~~~python
from sys import path
path.append("")

# start with the module import here
import mymodule
~~~

dpgeorge, 2021-12-17:
Fixed by e0bf4611c3a8b23b3c52e6a7804aac341ac3a87d

closed 2021-12-17 by dpgeorge