micropython/micropython #6422: STM32 HAL Driver Update

ckilimci, 2020-09-09:
Hello all,

Recently, STMicroelectronics updated STM32F4 HAL driver [1]. Some of the APIs are changed, like CAN. Do you plan to update included sources listed here?

[1] https://www.st.com/resource/en/user_manual/dm00105879-description-of-stm32f4-hal-and-ll-drivers-stmicroelectronics.pdf 

Sincerely,
Caglar

dpgeorge, 2020-09-17:
> Some of the APIs are changed, like CAN.

MicroPython provides a public *Python* API, and the underlying C API from the HAL doesn't really matter so much, as long as it functions correctly.  So unless there are bug fixes, it's a lot of effort to update the HAL for little benefit to the Python user.

In the end the "HAL" is really the set of registers for the peripherals, and that never changes.

Do you have other reasons to want to update the HAL?

BTW, a good source for the HAL is now GitHub: https://github.com/STMicroelectronics/STM32CubeF4

ckilimci, 2020-09-17:
First of all, thank you for your time and reply. 

I agree, there are not much benefit for the regular Python users but in the CAN legacy driver, there is a note that says:
`This HAL CAN driver is deprecated, it contains some CAN Tx/Rx FIFO management limitations. Another HAL CAN driver version has been designed with new API's, to fix these limitations`

>Do you have other reasons to want to update the HAL?

The main reason to examine updating HAL is, I am having an issue for receiving CAN messages if there is a burst of many frames, then I could not read all of them. Maybe, we can solve or improve just using interrupt mode of CAN driver. Do you think, updating HAL driver helps that, too? 

>BTW, a good source for the HAL is now GitHub: https://github.com/STMicroelectronics/STM32CubeF4

Thank you for this.

dpgeorge, 2020-09-17:
> The main reason to examine updating HAL is, I am having an issue for receiving CAN messages if there is a burst of many frames, then I could not read all of them. Maybe, we can solve or improve just using interrupt mode of CAN driver. Do you think, updating HAL driver helps that, too?

It's hard to say without looking into it in detail.  If the new HAL is substantially different to the old one then it'd be a fair bit of work to update the Python bindings to work with the new HAL and still give the same Python API to the user.  And that may not fix the problem you see.

ckilimci, 2020-09-17:
Thank you for precious comments.

I also saw the RFC you have created [6125](https://github.com/micropython/micropython/pull/6125) to enhance uevent module. As I understand, even that will be merged, I need to make changes to use interrupt mode of CAN driver, right?