micropython/micropython #6423: esp32/bluetooth: commit 126f972 breaks ble.active(True)

mbuesch, 2020-09-09:
126f972c34 breaks ``ble.active(True)``

With that commit applied, ``ble.active(True)`` always fails with ``OSError: [Errno 110] ETIMEDOUT``

mbuesch, 2020-09-09:
With the direct predecessor commit 311b8519af75be9fe4215b5ee8f77e91d089d5df bt works fine.

jimmo, 2020-09-10:
@mbuesch Thanks for the report!

Which ESP32 board are you using? Is it IDF 3.0 or 4.0?

With the previous commit, how long does `ble.active(True)` -- i.e. is the problem that it's taking longer than two seconds (and therefore should we change `NIMBLE_STARTUP_TIMEOUT`) or is there some sort of deadlock/synchronisation issue going on with the timeout.

Would it be possible (with and without 126f972 applied) to enable DEBUG_printf at the top of modbluetooth_nimble.c and paste the log output. Thanks

sactre, 2020-09-18:
I have the same problem after updating from tag 1.13 to master (42342fa)
`File "domain/services/ble_factory_mode_service.py", line 30, in start
OSError: [Errno 110] ETIMEDOUT`

sactre, 2020-09-21:
If I remove the commit 126f972, the bluetooth works for me, but when I try to connect to the wifi it gives me an error.
I (38184) wifi: STA_DISCONNECTED, reason:15
I (41344) wifi: STA_DISCONNECTED, reason:205
I (53974) wifi: STA_DISCONNECTED, reason:2

I change the beacon every 2 seconds, before commit 311b8519af75be9fe4215b5ee8f77e91d089d5df this works perfectly and I can continue using the wifi without problems. But since MP is running on core0 I can no longer use the wifi at the same time that I do that intensive use of bluetooth.

I guess now I'm not giving core 0 enough time to handle the wifi connection. I don't know if because of the use of bluetooth or how I have the application set up with asyncio.

FalconBFR, 2020-10-26:
same issue here - posted on
https://forum.micropython.org/viewtopic.php?p=51819&sid=421bb6a1f2851a8f768f882ffb4b9c25#p51819

jsamch, 2021-02-09:
If that can help someone, I had the same **OSError: [Errno 110] ETIMEDOUT** error.

I run _MicroPython v1.14 on 2021-02-02; ESP32 module with ESP32_

To workaround it, I had to issue the following before calling the BLE.active(True):
```
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
```

I think that some internal setup is missing (a clock maybe ?).

By activating the Wifi the internal chip seems to up corrects it for BLE operation.

Simanas, 2021-03-17:
I am trying to run [ble_simple_peripheral.py](https://github.com/micropython/micropython/blob/master/examples/bluetooth/ble_simple_peripheral.py) and getting the same result. 
If I activate WLAN as @jsamch recommends it passes, but falls short after that with OSError: -18 on **BLE.gap_advertise(...)** call.

Anybody has any idea how to fix it?  As untill this is not fixed BLE is practically unavailable for us to use.

Simanas, 2021-03-17:
Did some more digging and here's what I have found. 

It works perfectly fine as expected with ESP-IDF v4.x latest esp32-20210317-unstable-v1.14-120-g42cf77f48.bin, no need for wlan activation.

As per OSError: -18, on BLE.gap_advertise that should probably be a separate issue as it looks like there is some maximum length of data that can not be exceed on adv_data parameter. If you exceed it you get this error.

mbuesch, 2021-03-17:
The commit that caused this problem has long been identified.
In my opinion that commit should be reverted to fix this regression, as long as there is no real fix available.
That's the usual way to handle regressions.

Simanas, 2021-03-18:
Yes, something needs to be done about this... now with ESP-IDF v4.x I am unable to use WLAN features which are essential for my project. 

Though method proposed by @jsamch really does work!

Gaudon, 2024-03-13:
I frequently encounter this on multiple picoW microcontrollers and the only way to seemingly correct it is to power down the chip and reboot (sometimes multiple times) until it succeeds. A soft reset will never succeed in my experience.

dpgeorge, 2024-03-13:
> I frequently encounter this on multiple picoW microcontrollers

@Gaudon can you please provide full details on how you are able to trigger this issue: the hardware used (I assume a standard, unmodified Pico W board), the version of MicroPython (do `import sys` then `sys.version`) and the exact code that triggers the issue.  Then we can try to reproduce the issue and fix it.

Gaudon, 2024-03-13:
> @Gaudon can you please provide full details on how you are able to trigger this issue: the hardware used (I assume a standard, unmodified Pico W board), the version of MicroPython (do `import sys` [...]

I can certainly give it a shot. 

It's a 2022 PicoW (bought a few weeks back) running MicroPython. The issue happens even if I flash firmware to wipe the pico completely, reflash micropython and then load my project. 

It is however occasional and intermittent (fun!) and happens on both chips (My project uses 2 pico w's that communicate via serial) and each pico is running different code. 

The project in question is recently open sourced and can be found below:

[Zephyr Cycling - Smart Fan](https://github.com/Gaudon/zephyr-cycling)

The files of interest will be:

[bluetooth_receive_service.py](https://github.com/Gaudon/zephyr-cycling/blob/master/services/bluetooth_receive_service.py)

[bluetooth_transmit_service.py](https://github.com/Gaudon/zephyr-cycling/blob/master/services/bluetooth_transmit_service.py)

The first chip (primary mode) uses the first, the second chip (most recently had this issue, I'm dealing with it right now intermittently) runs the transmit service.

```
[ServiceLocator] : Service Registered [ConfigService]
**DEBUG:root:[SYSTEM] : Version - 3.4.0; MicroPython v1.22.1 on 2024-01-05**
[ServiceLocator] : Service Registered [InputService]
[ServiceLocator] : Service Registered [UartReceiveService]
[ServiceLocator] : Service Registered [BluetoothTransmitService]
INFO:root:[SYSTEM] : Initialized - Mode [SECONDARY]
DEBUG:root:[SYSTEM] : Memory - 52864 of 139072 bytes used (38%).
DEBUG:root:[BluetoothTransmitService] : State Changed - BROADCASTING
DEBUG:root:[SYSTEM] : Memory - 54176 of 137760 bytes used (39%).
```

This was a successful boot I just performed with the logging you requested! hah

dpgeorge, 2024-03-14:
@Gaudon thanks for the extra info.

> and the break/error code will point to the ble.active() line in the BluetoothTransmitService class

Can you please state the exact exception that is raised, the exact error message that's printed to the REPL (including traceback lines).

> I do not know if this is somehow related to it being connected to the debugger.

What do you mean by "connected to the debugger"?  Do you mean just a USB (or UART?) serial port with REPL, or do you have a SWD debugger attached separately?

> It is however occasional and intermittent (fun!) and happens on both chips (My project uses 2 pico w's that communicate via serial) and each pico is running slightly different code.

Unfortunately there's not enough information here for me to try and reproduce the issue.  Can you please provide instructions on how to deploy your application (which files go where on the Pico W), how to set up the two separate devices, and how to run them.

Gaudon, 2024-03-14:
@dpgeorge I rebooted the chip ~12 times, it's now hanging at the bluetooth service on startup.

[image]

Gaudon, 2024-03-14:
```
Traceback (most recent call last):
  File "<stdin>", line 73, in <module>
  File "asyncio/core.py", line 1, in run
  File "asyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 39, in main
  File "services/bluetooth_receive_service.py", line 47, in __init__
KeyboardInterrupt: 
```

I interrupted it after it was hung there for several hours (the board was unresponsive during this time).

bluetooth_receive_service.py
```
# Bluetooth
self.ble = bluetooth.BLE()
self.ble.active(True) <-- Line 47
```

Not particularly useful info here.

dpgeorge, 2024-03-15:
@Gaudon thanks for the additional details.  It looks like your issue is unrelated to the original issue of this ticket, because you don't see ETIMEDOUT raised from the `ble.active(True)` line.  So I will make a new issue for your case: #14097.

dpgeorge, 2024-03-15:
I'm going to close this issue because the original reports of the bug on esp32 are quite old now (3 years) and in the meantime:
- The IDF was updated to v5.
- BLE on esp32 now uses synchronous events.
- BLE on esp32 now has different core pinning behaviour.
- BLE on esp32 has had many bug fixes.
- All BLE multitests work on esp32 without any issue running `ble.active(True)`.
- Many people are using BLE on esp32 without any issue reported about `ble.active(True)`.

I suspect the issue may have been with IDF v3.

closed 2024-03-15 by dpgeorge