micropython/micropython #6424: Unix port can import uasyncio from REPL but not from file

mogenson, 2020-09-09:
I'm using the `dev` variant of the unix port from the 1.13 tag. This variant should have a frozen `uasyncio` module.

Typing `import uasyncio` from the REPL works:
```
$ micropython-dev
MicroPython v1.13 on 2020-09-09; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import uasyncio
>>>
```

Including the statement in a python file triggers an `ImportError`
```
$ echo 'import uasyncio' > test.py
$ micropython-dev test.py
Traceback (most recent call last):
  File "test.py", line 1, in <module>
ImportError: no module named 'uasyncio'
```

dlech, 2020-09-09:
Duplicate of #6419?

closed 2020-09-10 by mogenson

hoihu, 2021-05-16:
It still seems to be an issue, as of current master (1.15+).