micropython/micropython #6440: Implement exit() for ./micropython  shell

nanjekyejoannah, 2020-09-13:
I know that currently `CTRL-D` exits the shell. 

For consistency with other python shells i.e CPython, let us implement `exit()` as an option to also exit the shell.

If this sounds good, i will track this for myself so that I can implement this as part of google season of docs.

jimmo, 2020-09-14:
Consistency is good, although I must admit I've never found myself needing this.

However, an interesting note about builtin `exit()` in CPython is that it's provided by site.py as a shortcut for `sys.exit(0)` (which is already supported in MicroPython). So perhaps the broader feature here is "site.py for the Unix port"?

ref: https://github.com/python/cpython/blob/master/Lib/site.py#L373

In some sense `boot.py` is the same concept for the bare-metal ports.

dpgeorge, 2020-09-15:
Regarding site.py, see also #1741