micropython/micropython #6443: Please consider DOC entry on compatible GCC compilers

Willi-Villaton, 2020-09-14:
Hey   guys, 

so im currently working with micropython on a PIC controller. Because of problems with C extensions i tried to compile the stand-alone Windows-Port("make" in cmd on micropython/ports/windows). After two days of trying to find the error i recognised that my colleague was running the same configs on an older and different version of gcc. So we researched this issue a bit and came to the conclusion that **MinGW does not compile it whilst Octave GNU does**. We tested this for Octave Gcc Version 7.4.0 and 9.2.0 and MinGW Gcc Version 8.2.0 and 9.2.0. Then we compared the latest compiler versions (9.2.0) and found that those two versions differ in size and compile style.
[image]

But sadly we are in a bit of a rush with our project and dont have the time to do a full research. But it would be nice if you could implement some kind of list on the webpage or in the Docs in wich all suitable compilers are named. Thank you :)

Greetings!

stinos, 2020-09-14:
> conclusion that MinGW does not compile it whilst Octave GNU does

That's not a very accurate description: what you call 'octave gcc', i.e. the gcc which is distributed with Octave, is actually still just 'mingw gcc' i.e. a build of gcc which runs on Windows. The file size of the compiler is not relevant here, however it would be very helpful if you could run `gcc --version` for both versions and post the output, and  Additionally it would be very interesting to know exactly which errors you encounter: we cannot really change the documentation without knowing what is wrong. Also note that the Appveyor and Travis builds can succesfully build the Windows port using a variety of gcc versions, both MSYS2 and cross compiler versions, so we don't mention version numbers explcitly.

Willi-Villaton, 2020-09-14:
Hey, that was fast :D
What you requested (gcc --version) is what is was trying to point out with Octave Gcc and MinGW Gcc.
Used gcc versions (not MinGW or Octave Version) are [MinGW]: 8.2.0, 9.2.0 and [Octave]: 7.4.0, 9.2.0.
**MinGW Version recent(updated today to check with the recent gcc version) with GCC Version 9.2.0:** 
[image]
[image]
**Octave  Version 5.2 with GCC Version 9.2.0:**
[image]
[image]

User Variable Path in Example:
(I swap them to use the one needed)
[image]

For the second point:
This error comes up for every direct definition of "__USE_MINGW_ANSI_STDIO"
[image]
We then tried to take out the -Werror flag inside the makefile to only treat them as a warning but there are more problems exept those while trying to build the QSTR.

Willi-Villaton, 2020-09-14:
Addition:
I just reproduced the errors without -Werror in the ports/windows/Makefile for completion.

Firstly there are all warnings because of the direct definition of "__USE_MINGW_ANSI_STDIO"
As seen above

Secondly there are a whole bunch of other warnings which also address the declaration of use_mingw_ansi_stdio but with a pragma message:
[image]

And finally there is this error wich occures through the definition of EWOULDBLOCK / MP_EWOULDBLOCK. Somethings undefined here:
[image]
This also occures for every time EWOULDBLOCK is used/referenced.

stinos, 2020-09-14:
Thanks for the information! I didn't check it in detail yet, but unless I'm mistaken the problem is that the mingw gcc version is 'mingw32' and not 'mingw-w64' (which is a more common fork of mingw32 and also the one which comes with Octave). Those names are super confusing, but the point is that MicroPython only supports mingw-w64 and not mingw32. This is briefly hinted in the Windows README.md file, though it could probably be made more clear. Moreover the information in the README is a bit out of date: though it specifically mentions 'you may need to disable -Werror' it still sounds as if it might work, but probably since it was written new features were added and it now doesn't compile anymore at all, as you figured.

Can you confirm that's the problem? Where does your mingw distribution come from?

Willi-Villaton, 2020-09-17:
Hey, that actually seems to be the problem. 
My MinGW distribution is from the official site which is indeed the 32bit version. (https://osdn.net/projects/mingw/releases/)
I would recommend adding this information to the "Getting Started" on the Homepage. This would be really helpful for beginners to not get knocked over with weird error messages after only doing what is hinted in various getting started tutorials.
Thank you for the fast support and the solution :)

stinos, 2020-09-17:
> I would recommend adding this information to the Docs on the Homepage. 

You mean on http://docs.micropython.org? Thing is that's mostly about MicroPython, the language, and has no build instructions (I think). The go-to place for build instrcutions is really the README.md for each port. However it would help if that was made much clearer in the main README.md instead of buried in a sentence halfway.

Willi-Villaton, 2020-09-17:
> However it would help if that was made much clearer in the main README.md instead of buried in a sentence halfway
Yep thats right. In any case a clear hint would be great and save a lot of detective work in the unfortunate case of a wrong compiler.
And maybe it would be worth considering a "Getting Started" Page on http://docs.micropython.org to make things easier for beginners in micropython like myself. =)

stinos, 2020-09-21:
Little update: there's been some discussion on this in Slack. Summary: everyone agrees that the build instructions should be easy to find from the main README; it serves as the landing page for the project so the instructions for 'how do I start this thing' should be at most 1 click away. It's not too hard to add such links, but because there's an ongoing Season of Docs project for MicroPython, and as part of that build instructions might be moved around so it doesn't make much sense of starting parallel work to get this sorted out, yet.

stinos, 2020-09-22:
See #6469

closed 2020-09-25 by Willi-Villaton