micropython/micropython #6455: ENOMEM when using SSL connections - gc.mem_free() > 80000

faustomilletari, 2020-09-17:
Dear micropython team,

i am encountering this odd situation where i cannot establish any non-ssl connection unless i do it right at the beginning of my script. I am currently using a espressif EPS32 module (using a dev board ESP32 that i bought from amazon). I have tried with both 1.12 and 1.13 firmwares.

What i am observing is that, whenever I try a SSL connection in the middle of my script, the connection fails with os.error ENOMEM. If i realize the same connection at the beginning of the script, it succeeds. 

I know ENOMEM is a memory exhausted error, therefore i have run gc.collect() and retried. the result is the same. I should also mention that the gc.mem_free() command returns always a number > 80k in my case. That's not much different than what i get right at the beginning of my script.

I have also visualised the heap fragmentation, to see if maybe my heap was too fragmented. But it turns out that i always have most of the heap free for allocation in a contiguous way.

I have found another post that was mentioning that the problem might not be the heap. it might be some other memory that is exhausted. I don't remember were i have read this, but any clarification will help.

I really would appreciate any help. I would be already very happy if i could simply be able to retrieve some information about my memory usage so that i can identify the cause of the ENOMEM and maybe reduce memory allocation/consumption elsewhere in the program in order to be able to fit the SSL sockets.

faustomilletari, 2020-09-17:
Another thing that i noticed is that gc.mem_free() does not seem to be reliable. If i open a websocket over SSL and then close it, it seems that there is almost no difference between memory allocation before, while and after websocket was opened.

```
>>> gc.collect()
>>> gc.mem_free()
98736
>>> websocket = uwebsockets.client.connect('wss://echo.websocket.org')
>>> gc.collect()
>>> gc.mem_free()
98784
>>> websocket.close()
>>> gc.collect()
>>> gc.mem_free()
98864
>>> websocket = None
>>> gc.collect()
>>> gc.mem_free()
98848
>>> del websocket
>>> gc.collect()
>>> gc.mem_free()
98912
```

tve, 2020-09-17:
The problem is that the C heap, not the Python heap is full. Make sure you promptly close any sockets that you no longer use so the memory gets freed. If you must have multiple TLS sockets open, use a module with PSRAM, or compile with bluetooth disabled if you don't use it, or reduce the memory allocated to python in main.c.

tve, 2020-09-17:
If you want to try, you can download my BLE-less firmware at https://tve.s3.amazonaws.com/mpy-builds/20200917-TOOTHLESS_OTA-v1.12-tve2/index.html. You have to erase the flash if you get a filesystem error at boot.

faustomilletari, 2020-09-17:
Unfortunately I am using also Bluetooth to allow my users to configure the WiFi connection in a touch less manner :(

In terms of sockets, I just need exactly one... any way I can monitor current usage of c heap from python or else? I have no idea what else might be filling it up apart from the BLE stack.

I thank you very much for your help. This clarifies things quite a bit already... I just hope I can debug more and see what’s consuming memory.

tve, 2020-09-17:
Open the TLS socket before you start BLE, if you can...
Subtract 20*1024 in https://github.com/micropython/micropython/blob/master/ports/esp32/main.c#L103

faustomilletari, 2020-09-17:
Thanks a lot! I will try in this way or maybe shift to esp32-wrover if you advise it as a possible solution having more ram! 

Thanks

closed 2020-09-17 by faustomilletari