micropython/micropython #6456: for fname in os.listdir(path): last item wrong bug

emard, 2020-09-18:
At ESP32 micropython v1.13 appears a bug which was not on v1.12
Last item from listdir() is corrupted and becomes a wrong value.

For example I have uftpd.py server, slightly modified/bugfixed for my purpose
but listdir() is the same. I enter some directory with 20-30 files and do
ftp> ls
Last item is wrong

Bug appears here
https://github.com/emard/esp32ecp5/blob/master/new/uftpd.py#L84

for fname in uos.listdir(path):
  if I print(fname) i see correct value but when it is passed in
  self.make_description(path, fname, full))
  there if I print(fname) its already wrong
  fname typically gets changed to directory name (last part path) but it shouldnt

I think its a bug in micropython v1.13

dpgeorge, 2020-09-18:
Are you using FAT FS or littlefs?  Run the following script to find out:
```python
"""
Dump info about a filesystem.
"""

import sys, struct

def pr(addr, label, data):
    if type(data) == bytes or type(data) == bytearray:
        data = ':'.join('%02x' % b for b in data)
    print('%04x  %-32s %s' % (addr, label, data))

def decode_bootsec_fat(b):
    print('FAT filesystem')
    pr(0, 'jump', b[0:3])
    pr(3, 'OEM name', str(b[3:11], 'ascii'))
    pr(11, 'sector size (bytes)', struct.unpack_from('<H', b, 11)[0])
    pr(13, 'cluster size (sectors)', struct.unpack_from('<B', b, 13)[0])
    pr(14, 'reserved area (sectors)', struct.unpack_from('<H', b, 14)[0])
    pr(16, 'number of FATs', struct.unpack_from('<B', b, 16)[0])
    pr(17, 'size of root dir area', struct.unpack_from('<H', b, 17)[0])
    pr(19, 'volume size (sectors)', struct.unpack_from('<H', b, 19)[0])
    pr(21, 'media descsriptor', hex(struct.unpack_from('<B', b, 21)[0]))
    pr(22, 'FAT size (sectors)', struct.unpack_from('<H', b, 22)[0])
    pr(24, 'track size (sectors)', struct.unpack_from('<H', b, 24)[0])
    pr(26, 'number of heads', struct.unpack_from('<H', b, 26)[0])
    pr(28, 'volume offset (sectors)', struct.unpack_from('<L', b, 28)[0])
    pr(32, 'volume size (32-bit) (sectors)', struct.unpack_from('<I', b, 32)[0])
    pr(36, 'physical drive number', struct.unpack_from('<B', b, 36)[0])
    pr(37, 'error flag', hex(struct.unpack_from('<B', b, 37)[0]))
    pr(38, 'extended boot signature', hex(struct.unpack_from('<B', b, 38)[0]))
    pr(39, 'volume serial number', hex(struct.unpack_from('<L', b, 39)[0]))
    pr(43, 'volume label', str(b[43:51], 'ascii'))
    pr(54, 'filesystem type', str(b[54:62], 'ascii'))
    pr(510, 'signature', hex(struct.unpack_from('<H', b, 510)[0]))

def decode_bootsec_lfs1(b):
    print("Littlefs v1 filesystem")
    pr(24, "type", struct.unpack_from("<I", b, 24)[0])
    pr(25, "elen", struct.unpack_from("<I", b, 25)[0])
    pr(26, "alen", struct.unpack_from("<I", b, 26)[0])
    pr(27, "nlen", struct.unpack_from("<I", b, 27)[0])
    pr(28, "block_size", struct.unpack_from("<I", b, 28)[0])
    pr(32, "block_count", struct.unpack_from("<I", b, 32)[0])
    pr(36, "version", hex(struct.unpack_from("<I", b, 36)[0]))
    pr(40, "magic", str(b[40:40 + 8], 'ascii'))

def decode_bootsec_lfs2(b):
    print("Littlefs v2 filesystem")
    pr(8, "magic", str(b[8:8 + 8], "ascii"))
    pr(20, "version", hex(struct.unpack_from("<I", b, 20)[0]))
    pr(24, "block_size", struct.unpack_from("<I", b, 24)[0])
    pr(28, "block_count", struct.unpack_from("<I", b, 28)[0])
    pr(32, "name_max", struct.unpack_from("<I", b, 32)[0])
    pr(36, "file_max", struct.unpack_from("<I", b, 36)[0])
    pr(40, "attr_max", struct.unpack_from("<I", b, 40)[0])

def decode_bootsec(b):
    if b[40:48] == b"littlefs":
        decode_bootsec_lfs1(b)
    elif b[8:16] == b"littlefs":
        decode_bootsec_lfs2(b)
    else:
        decode_bootsec_fat(b)

def main():
    if sys.platform == 'pyboard':
        import pyb
        #sd = pyb.SDCard()
        sd = pyb.Flash(start=0)
        bootsec = bytearray(512)
        sd.readblocks(0, bootsec)
    elif sys.platform in ('esp8266', 'esp32'):
        import esp
        bootsec = bytearray(512)
        esp.flash_read(esp.flash_user_start(), bootsec)
    else:
        with open(sys.argv[1], 'rb') as f:
            bootsec = f.read(512)
    decode_bootsec(bootsec)

main()
```

robert-hh, 2020-09-18:
+1 for that nice script.

dpgeorge, 2020-09-18:
> +1 for that nice script.

Yeah, I find it very useful!  If you have suggestions for where things like this could go, I can publish it (eg micropython-lib, a new repo, or a directory here).

robert-hh, 2020-09-18:
Haveing that "officially" published would be fine. I was wondering myself where to put it. A good place would be micropython/tools or similar. micropython-lib was not my favorite. I'm unclear about the state, since I do not know who is maintaining it.
Where did you place th new mpr tool?

dpgeorge, 2020-09-18:
> A good place would be micropython/tools or similar.

I have a local tools directory with quite a few of these things.  Since they are rather ad-hoc (and not intended to be put on PyPI for example) might be better suited to a new repo.

> Where did you place th new mpr tool?

I did not merge that yet, it's still in a PR... probably that one belongs in this repo under `tools/`, alongside `pyboard.py`.

(Sorry to OP, I didn't intend to derail this issue, so let's not discuss it more here.)

emard, 2020-09-18:
In my case, I see this problem on a FAT filesystem of a SD card.
I have there a lot of files and directories but usually not too many
20-30 per directory. And it happens with this particular type of
code (uftpd). If I have my own SD card browser, it's a different
coding style and then it doesn't happen

I can mount the same SD as FAT at linux too if that matters

robert-hh, 2020-09-18:
@emard If it only happens with uftpd, it' not in @dpgeorge's playground. It's more my job at that repo:https://github.com/robert-hh/FTP-Server-for-ESP8266-ESP32-and-PYBD

dpgeorge, 2020-09-18:
> In my case, I see this problem on a FAT filesystem of a SD card.

So the problem is unlikely to be with the filesystem format, but rather with more low-level things, like a change to the SDCard driver on esp32.

> if it only happens with uftpd, it' not in @dpgeorge's playground. It's more my job at that repo

True.  But if exactly the same code and hardware worked on v1.12... hard to say.

robert-hh, 2020-09-18:
> But if exactly the same code and hardware worked on v1.12... hard to say.
Sure. I'll look into that too.

emard, 2020-09-18:
Bug is even more strange because it doesn't happen
to all directories but only on some.
I tried to fill some directory
tree wth 15-20 new files and 2-3 new dirs using v1.13 and it looks good.

dpgeorge, 2020-09-18:
If the esp32 module has SPIRAM then that could explain it.

emard, 2020-09-18:
Module is WROOM, no SPIRAM

emard, 2020-09-18:
I have second SD card (8GB) where it doesnt happen.
On the first SD (64GB) where it happens, I have 3 MSDOS paritition,
first is FAT 8GB, second is linux ext2 and third is linux swap I think.
ESP32 normally sees only FAT 8GB

emard, 2020-09-18:
the most files at problematic FAT were written using v1.12 and uftpd.

I have put SD card in linux PC, copied all subdirectories,
deleted them from SD and wriitten back from the PC copy

I wouldnt say its fixed so easy but at least the strange bug doesn't
appear anymore.

robert-hh, 2020-09-18:
I can confirm that there is an inconsistency, in that directories are not properly listed, depending on the command used for listing. In command line ftp (linux version):
nlist   lists all files and directories
dir or ls lists files, but no directories
rstat * lisst files, but no directories

Filezilla lists a sd(?) directory instead of the real ones.

But I have a possible source:

The program fails at  line 100 of uftpd.py:
`tm = localtime(stat[7])`
with the error: `OverflowError: overflow converting long int to machine word`
stat[7] being 3666235282

with stat[7]

robert-hh, 2020-09-18:
I've patched uftpd.py to replace the file mod time with the actual time if an overflow occurs. Alternatively the date could be left off from the display. But I do not know how ftp clients like filezilla then deal with it..

robert-hh, 2020-09-20:
I've ran into that problem again. ESP8266, today's build, littlefs, a file created just now by uploading with uftpd.py had strange creation times.
```
>>> uos.stat("test.py")
(32768, 0, 0, 0, 0, 0, 1563, 3348282609, 3348282609, 3348282609)
>>> time.localtime(uos.stat("test.py")[7])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word
```
And the respective code in uftpd.py harmless. It may be that sometime localtime() is not properly initialized.

dpgeorge, 2021-04-14:
Hopefully this is fixed.  Feel free to reopen the issue if not.

emard, 2021-04-14:
I'd like to check for esp32,
after which binary release is this issue fixed?

dpgeorge, 2021-04-14:
It should be fixed in v1.14.  You probably also need the latest uftpd.py.

emard, 2021-04-14:
I tried official release 1.14 it was not fixed, same as 1.13.
Latest is 1.12 that doesn't have this bug.

It is not really fixable in any new uftpd.py (it could only
hide the bug). It's about the list which returns value wrong

I can try some daily build 1.14-something if fix came

emard, 2021-04-14:
I tried esp32-20210414-unstable-v1.14-161-g2ac09c269.bin
and bug is still there.

robert-hh, 2021-04-14:
What exactly did you try. I did the same just now for  a few files, and the file creation dates are fine.

emard, 2021-04-14:
I have modified uftpd.py for my application
https://github.com/emard/esp32ecp5/blob/master/uftpd.py

there are "site" command additions to simplify
mounting/unmounting SD card and other similar things
few bugs fixed like file open is "rb" for binary, not "r" for ascii

but listdir() part is the same, unchanged, but exhibits the bug

the latest uftpd.py doesn't exhibit the bug but I think it
doesn't mean it's gone. If I simply place just listdir() command
on v1.13 or v1.14 it also doesn't exhibit bug

emard, 2021-04-14:
the bug: listdir() returns less items that
really exist on SD card. Last file is typically
repeated directory name which doesn't exist.

For example SD card is mounted on "/sd"
Then listdir at /sd lists few files/dirs and sd
is repeated but it doesn't exist really

> cd sd
> ls
file1
file2
dir1
dir2
sd  <---- this doesn't exist on SD card, wrong listed
>

robert-hh, 2021-04-14:
If there are bugs please raise an issue. The SITE command was added recently.
Do you see this issue of the wrong last entry only with the SD card?

emard, 2021-04-14:
no, last "SD" entry is not the only wrong thing,
the many files from SD directory are missing.
How many missing? one or more, it varies from
SD and its content.

typical appearance is that at least one file/dir is
missing and always last listed item is repeated sd
item. that's how I immediately recognize the bug at files
list

I thought that issue has already been raised, only
I don't have minimal code that shows the bug, only
on real application like my uftpd.py it appears

My easiest way is to take new uftpd.py, manually
port my changes around "site" command and I will
fix my problem, but bug will continue to exist and
appear somewhere else

robert-hh, 2021-04-14:
I just tried that with an ESP32 and SD card, and it worked fine. I see no problems. When using ftp command line for acess, are the files missing both for the ls and nlist command?

P.S.: I just changed uftpd to use binary mode file open. As said, this version also has a "SITE" command, suggested by another user. Maybe different to your implementation.

emard, 2021-04-14:
site generic command is ok. I just added
some useful site command shortcuts that reduce typing
for my specific use, but that's nothing really important for ftp function

I placed print

      for fname in uos.listdir(path):
        print(fname)
        data_client.sendall(self.make_description(path, fname, full))

and it shows similar as "ls" but without last "sd" directory, see

lftp 192.168.48.181:/> cd sd
cd ok, cwd=/sd
lftp 192.168.48.181:/sd> ls
-rw-r--r-- 1 owner group      45267 Mar 25  2008 KERNEL.SYS
-rw-r--r-- 1 owner group      64919 Mar 25  2008 COMMAND.COM
-rw-r--r-- 1 owner group       4607 Mar  3  2008 INVADERS.COM
-rw-r--r-- 1 owner group       4848 Oct  1  2002 CTMOUSE.EXE
-rw-r--r-- 1 owner group     524288 Jan  1 00:03 kick.rom
-rw-r--r-- 1 owner group      69548 Jan  1 00:03 OSD_CA01.sys
drwxr-xr-x 1 owner group          0 Sep 18  2020 amiga
drwxr-xr-x 1 owner group          0 Sep 18  2020 colecovision
drwxr-xr-x 1 owner group          0 Sep 18  2020 hdl4fpga
drwxr-xr-x 1 owner group          0 Jan  1 00:00 sd
lftp 192.168.48.181:/sd>

on the console, print shows this

>>>
>>>
>>> KERNEL.SYS
COMMAND.COM
INVADERS.COM
CTMOUSE.EXE
kick.rom
OSD_CA01.sys
amiga
colecovision
hdl4fpga
linux

there are much more items on SD card unlisted

emard, 2021-04-14:
the full content is this
KERNEL.SYS
COMMAND.COM
INVADERS.COM
CTMOUSE.EXE
kick.rom
OSD_CA01.sys
amiga
colecovision
hdl4fpga
linux
m68k
mp3
msx
nes
next186
orao
ql
scope
sega_sg1000
rootfs.cpio.uboot
uImage
uboot.env.defunct
dtb
slides
ti99_2
ti99_4a
vic20
zxspectrum
snes
rtc
myamoto
apple2
accel.wav
speak
c64

On 4/14/21, D EMARD ***@***.***> wrote:
> site generic command is ok. I just added
> some useful site command shortcuts that reduce typing
> for my specific use, but that's nothing really important for ftp function
>
> I placed print
>
>       for [...]

emard, 2021-04-14:
I just checked,
"nlist" shows all files but "ls" shows reduced file list

On 4/14/21, D EMARD ***@***.***> wrote:
> the full content is this
> KERNEL.SYS
> COMMAND.COM
> INVADERS.COM
> CTMOUSE.EXE
> kick.rom
> OSD_CA01.sys
> amiga
> colecovision
> hdl4fpga
> linux
> m68k
> mp3
> msx
> nes
> next186
> orao
> ql
> scope
> sega_sg1000
> rootfs.cpio.uboot [...]

emard, 2021-04-15:
Maybe listall() some error happens in
make_description() and it silently gets catched by "try",
passing execution to "except" where it is causing
"sd" to be reported

emard, 2021-04-15:
yes it is unseen error in try/except
I captured the error it happens when trying to
make_description() for "linux" directory because
linux directory has something different than other directories
which does integer overflow

Error is in line
   tm = localtime(stat[7])

colecovision
drwxr-xr-x 1 owner group          0 Sep 18  2020 colecovision

hdl4fpga
drwxr-xr-x 1 owner group          0 Sep 18  2020 hdl4fpga

linux
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lsbug.py", line 40, in <module>
  File "lsbug.py", line 38, in show
  File "lsbug.py", line 22, in make_description
OverflowError: overflow converting long int to machine word
>>>

emard, 2021-04-15:
Here I print problematic integer before the error
    print(stat[7])

colecovision
653738842
drwxr-xr-x 1 owner group          0 Sep 18  2020 colecovision

hdl4fpga
653738844
drwxr-xr-x 1 owner group          0 Sep 18  2020 hdl4fpga

linux
3663815296
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lsbug.py", line 41, in <module>
  File "lsbug.py", line 39, in show
  File "lsbug.py", line 23, in make_description
OverflowError: overflow converting long int to machine word

emard, 2021-04-15:
when same code is running from v1.12 bug doesn't appear
because this integer value for directory "linux" is different
at v1.14 it is 3663815296 and error happens
at v1.12 it is -631152000 and no error

colecovision
653738842
drwxr-xr-x 1 owner group          0 Sep 18  2020 colecovision

hdl4fpga
653738844
drwxr-xr-x 1 owner group          0 Sep 18  2020 hdl4fpga

linux
-631152000
drwxr-xr-x 1 owner group          0 Jan  1  1980 linux

m68k
653738848
drwxr-xr-x 1 owner group          0 Sep 18  2020 m68k

robert-hh, 2021-04-15:
That was my suspicion. That's why I asked you to try nlist. The value of 3663815296 is exceeding the range of a short int. Can you try:

` tm = localtime(stat[7] & 0x7fffffff)`
That will give a wrong file modification time, but not fail. Still, this case should be catched by the `except OverflowError:` clause, whichj then only shows a reduced number of file information.

emard, 2021-04-15:
HI

the and-mask fix helps, the file is listed but with wrong date only

drwxr-xr-x 1 owner group          0 Jan 19  2048 linux

now it's explainable behaviour, try-except loop catches the overflow
error, terminatest the for-loop and thus cuts off remaining items.

I think try-except loop should be written differently or moved inside of
 for-loop that iterates fname to better handle eventual errors and not
cut off intems at the directory listing

emard, 2021-04-15:
my fix to handle the overflow and let the code
work the same for v1.12 and v1.14 is this:

      mtime = stat[7];
      while mtime >= 0x80000000:
        mtime -= 0x100000000
      tm = localtime(mtime)

robert-hh, 2021-04-15:
There is something else wrong, since OverflowError is handled in make_description() itself, returning an entry without dates.

emard, 2021-04-15:
latest uftpd.py handles this well
I had older uftpd.py which didn't have this try/except
but I think my integer arithmetic makes more gracefully
handling past dates with less code.
It lists actual value (date from 1980) better than
not listing the date, considering it something unknown

robert-hh, 2021-04-15:
Is the time displayed after subtracting 0x100000000 the proper one or a different wrong date?

emard, 2021-04-15:
yes after subtraction it showsproper date
same as when SD card is put in reader at
linux machine. Without RTC clock, default
VFAT dates normally start from 1980

robert-hh, 2021-04-15:
So I'll change that in uftpd.py. It may also be changed in time.localtime(). But that's another story. my code change is:
```
                tm = stat[7]
                tm = localtime(tm if tm < 0x80000000 else tm - 0x100000000)
```

emard, 2021-04-15:
seems ok to me

robert-hh, 2021-04-15:
I guess you can close the issue. 
P.S.: I looked where in localtime() the error was raised, but could not find the place.

closed 2021-04-16 by dpgeorge