micropython/micropython #6459: Clarify whether ubluetooth.BLE.irq handler is really an _interrupt_ handler

scy, 2020-09-19:
I’m currently tinkering around with the BLE support (I’m using the ESP32 IDF3 build of 1.13, fwiw) and noticed that some of the [examples](https://github.com/micropython/micropython/blob/42342fa3cb30e2eac56ceb1d21b4eb60a0f406f3/examples/bluetooth/ble_temperature_central.py#L87-L174) allocate memory etc. in the callback function passed to `irq()`.

So, apparently the [ISR rules](https://docs.micropython.org/en/latest/reference/isr_rules.html#isr-rules) don’t apply here, or at least not fully.

I know that the module is still experimental, but in order to do meaningful experiments with it, one has to know to some degree on what should be possible. 😉 Could you please clarify (ideally in the docs):

* whether, or to what degree, this callback is supposed to behave like an ISR
* which rules and best practices would apply when working with it and
* whether it’s important to return ASAP from the handler, in order to e.g. not miss scan results – or are they buffered somehow?

If the ISR rules largely or fully don’t apply to the handler function at all, considering to rename the `irq` function would be a good idea imho, because the name “IRQ” comes with some assumptions.

jimmo, 2020-09-20:
I agree this could be clearer, and will try and do an update to the docs soon.

The quick answer to your question is that it's a "soft" interrupt handler (i.e. it's invoked by the same mechanism as micropython.schedule) and therefore the same rules as other soft interrupt handlers apply (i.e. `pin.irq(..., hard=False)`). Namely: you can allocate, but you should still aim to return as quickly as possible as it prevents other scheduled callbacks from running (including from modbluetooth, but on ESP32 this also includes all pin interrupts, as it doesn't support hard=Trus).

In general I'd suggest do the minimum possible to figure out if you care about the given event (i.e. does this scan result contain the required advertising fields) but not stuff like "synchronously update the display with the new value written to the characteristic".

Events are (ring)buffered (and the 'rxbuf' config sets the size of this buffer).

FWIW, I've written an (very rough) asyncio-based wrapper around modbluetooth that hides most of this, but it needs `uevent` (or equivalent) support. Will send a PR (maybe to micropython-lib?) when that's available.

It's called `irq` for consistency with all the other objects you can set callbacks on, but yeah I agree the "irq" name is possibly being stretched a bit far here?

dpgeorge, 2021-04-14:
With synchronous BLE events, it's now much more like an IRQ.