micropython/micropython #6461: Enhance upip error message when installing a CPython package

aivarannamaa, 2020-09-20:
When a novice user accidentally tries to install a CPython package with upip, she gets a warning about SSL certificate not being validated but nothing really helpful for debugging the situation.

I propose that in cases where package name doesn't start with a known prefix (eg "micropython-" or "pycopy-"), upip should append "Are you sure this package is meant for MicroPython" to the original error message.

Here is a real issue from Thonny: https://github.com/thonny/thonny/issues/1330. (In this case micropip.py was used, but it might be good idea to implement error handling similarly in upip and micropip.py, https://github.com/peterhinch/micropython-samples/issues/17)

I can create a PR for this change.

peterhinch, 2020-09-20:
Note to maintainers.
I suggested to @aivarannamaa that he raised a PR against `upip` rather than against `micropip`; the latter is just a port of `upip` and I don't want its functionality to diverge. I'll port any changes that are accepted.

Re the actual change:
Do all PyPi packages for MP have the micropython prefix? Also up for grabs is whether pycopy should be supported because compatibility with official firmware is not guaranteed.

jrmoserbaltimore, 2021-03-25:
> Re the actual change:
> Do all PyPi packages for MP have the micropython prefix? Also up for grabs is whether pycopy should be supported because compatibility with official firmware is not guaranteed.

https://pypi.org/project/rgb-display/ accepts objects with a certain API but doesn't do anything in particular, should run on Pi Zero, Pi 4, Pi 3, Pi Pico.

jrmoserbaltimore, 2021-03-25:
Actually, have a look at programming languages:

```
Programming Language
Python :: 3
Python :: 3.4
Python :: 3.5
Python :: 3.6
Python :: 3.7
Python :: 3.8
Python :: 3.9
Python :: Implementation :: CPython
Python :: Implementation :: MicroPython
```

MicroPython.  Can we detect *that*?

jonnor, 2024-09-13:
upip is gone - replaced by mip.  See https://docs.micropython.org/en/latest/reference/packages.html
So this is no longer relevant.

closed 2024-09-13 by jonnor