micropython/micropython #6462: ESP32: Best Way for fast and correctly file upload

maxpromer, 2020-09-20:
Now, i develop ide for MicroPython, i needs how to fast and correctly file upload ? Now, i use Raw REPL mode for `open` `f.write()` and `f.close()` file. I found on ESP32 the `f.write()` can hold max about 500 Byte before call `f.close()`, i think because `f.write()` is write to RAM and will write to Flash when call `f.close()`. it **True** or **False** ? Next, i found after call `f.close()` we needs about 200mS wait for save to Flash and Can call `open` again.

OK, the main question: How to **fast and correctly file upload** ?

1. Is it possible ? , After call `f.close()` we can call `open` without wait time
2. Is it possible ? , MicroPython implement **Upload Mode** for upload MicroPython code without `open` `f.write()` and `f.close()`
3. Is it possible ? , Use some tool for convert file .mpy or .py to .bin and then use esptool for upload .bin into Flash

Thanks !

mirko, 2020-09-21:
Not sure if this helps but I'm using `mpfshell`[1] for multiple uploads of partially also rather large files.
It basically uploads a stub of basic functionality and then operates, using mpy's file/io routines on the lower level.

[1]https://github.com/wendlers/mpfshell

maxpromer, 2020-09-22:
> Not sure if this helps but I'm using `mpfshell`[1] for multiple uploads of partially also rather large files.
> It basically uploads a stub of basic functionality and then operates using mpy's file/io routines on the lower level.
> 
> [1]https://github.com/wendlers/mpfshell

Thank you for help. I will test.

i see `put()` method in `mpfexp.py`, it call `f.write()` without delay, i will learn how `mpfexp.py` it work with large file.

However, use f.write() we not know file write is correctly, Of most **checksum** is used for recheck correctly in low speed system. In high speed system, **MD5** or **SHA** use for recheck correctly ... and f.write() is not support checksum and MD5 and SHA.

jonnor, 2024-09-13:
The recommended approach now is to use mpremote, either as a command-line tool or Python module.

closed 2024-09-13 by jonnor