micropython/micropython #6464: Using clang-cl on Windows

gvanem, 2020-09-21:
I see that the sources have some tests in `__clang__`. But building with `clang-cl` (the MSVC-driver = `clang --driver-mode=cl`) 
seems unsupported. E.g. on trying to compile `examples/embedding/hello-embed.c`:
```
examples/embedding/hello-embed.c(68,6): error: conflicting types for 'mp_import_stat'
uint mp_import_stat(const char *path) {
     ^
.\py/lexer.h(194,18): note: previous declaration is here
mp_import_stat_t mp_import_stat(const char *path);
                 ^
```

Casting an `enum mp_import_stat_t` to an `uint`. What's going on?
I've never thought that was legal.

stinos, 2020-09-22:
That is not a cast, it's a different function signature/declaration (or whatever the correct term is) and I assume it's legal and also safe at runtime as long as the return value types have the same size, which they do since `mp_import_stat_t ` is an `enum`. 

That doesn't help you though.. But: the unix port builds with clang and unix' main.c also uses `uint mp_import_stat(const char *path)` so you might be using the wrong compiler flags and/or version or it's a bug in clang. I tested on https://godbolt.org/ and every C compiler I tried there accepts it (but it doesn';t have clang-cl..)

gvanem, 2020-09-22:
An enum is signed, `uint` is not (defined in clang's `<stddef.h>`). Size is the same, but that does not matter. 
But clang has no problem with `int mp_import_stat(const char *path)`.

Further, the `__clang__` in MicroPython seems to me to imply MacOS etc. Not windows. Because:
*  A builtin like `__i386__`  sets `MICROPY_NLR_X86=1` causing  `nlr_push()` to be called. Which crashes; fixed<br>by `-DMICROPY_NLR_SETJMP=1`.

stinos, 2020-09-22:
> An enum is signed, uint is not (defined in clang's <stddef.h>). Size is the same, but that does not matter.

Isn't an enum's signedness implementation-dependent, likewise for the actual size? For all we know all compilers make the enum unsigned because it has no values less than zero? Anyway, doesn't really matter as you say (I mean would be interesting to figure out what the standard has to say exactly about this), point is it doesn't compile. Are you sure you are using the exact same compiler flags as we use for clang on unix/osx?

> the __clang__ in MicroPython seems to me to imply MacOS etc

Yes, anything unix-like I guess, but indeed not Windows, so a patch for specific issues is always welcome.