micropython/micropython #6468: SSL/SNI issues

mladedav, 2020-09-22:
I am unable to connect to a TLS-enabled socket which is behind CloudFlare, i.e. I need to use SNI. I have found that the support is there in Micropython and the upstream axTLS but it does not work for me. I get a generic (and a bit wrong) error of OSError with -40, which turned out to probably not be the OSError code, but TLS error, but I cannot find it in the axTLS code.

I can connect to pages that do not use SNI and what's more, I can connect to them even if I use `server_hostname` with either valid or invalid domain name. When SNI is needed the same error is given regardless of `server_hostname` value.

I don't even know how to debug this behaviour and get anything more than just '-40'.

```
>>> addr = socket.getaddrinfo(host, 443)
>>> s = socket.socket()
>>> s.connect(addr)
>>> tls = ssl.wrap_socket(s, server_hostname=host)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -40
```

a0s, 2024-08-03:
I found probably the same error -40 when i tried to sent metrics to influx cloud with urequests.post.

dpgeorge, 2024-08-05:
Error -40 is likely the following from axtls:
```c
#define SSL_ALERT_HANDSHAKE_FAILURE             40
```
and that seems to be generated for the following case only:
```c
        case SSL_ERROR_NO_CIPHER:
            alert_num = SSL_ALERT_HANDSHAKE_FAILURE;
            break;
```
So it likely means that axtls doesn't support the ciphers requested by the server.

@a0s if you'd like to make progress on this issue, please post a fully-working piece of Python code that shows the issue.

jonnor, 2024-09-02:
Please see the following limitations about SSL on axltls/esp8266: https://docs.micropython.org/en/latest/esp8266/general.html#ssl-tls-limitations

There is unfortunately not much we can do to fix those. For using modern SSL, the esp8266 is best avoided.

closed 2024-09-02 by mladedav