micropython/micropython #6474: json dumps tests fail when CPython ujson gets imported

stinos, 2020-09-23:
Some package I installed in my CPython env has a dependency on ujson (the one from https://pypi.org/project/ujson/) so when running MicroPython tests like ujson_dumps.py and ujson_dumps_float.py, generation of the .exp files with CPython will import said ujson, but that defaults to using `,` and `:` as separator instead of the normal `, ` and `: ` with an extra space. As a result the test fail because MicroPython also uses the space.

I think it's worth fixing this in MicroPython to avoid it bothering others?

The tests now have
```
try:
    import ujson as json
except ImportError:
    try:
        import json
    except ImportError:
        print("SKIP")
        raise SystemExit
```
and changing that to
```
try:
    import json
except ImportError:
    try:
        import ujson as json
    except ImportError:
        print("SKIP")
        raise SystemExit
```
works, but I'm not sure it won't affect anything else.

dpgeorge, 2020-09-23:
Is there a way to run CPython with only the standard library, or something similar, so it doesn't pick up packages like this?

dpgeorge, 2020-09-23:
> and changing that to
> ...
> works, but I'm not sure it won't affect anything else.

That's not ideal because the unix port may have `json` installed in its path, and then the test is testing `json` instead of the built-in `ujson`.

stinos, 2020-09-23:
> Is there a way to run CPython with only the standard library, or something similar, so it doesn't pick up packages like this?

Haven't found such thing (yets); there's a `-I` which 'isolates' but only from user packages and environment variables. Problem here is the module is in site-packages, which is part of the deafult implementation-defined module search path and the only way I know to alter that is removing it manually from sys.path.

dlech, 2020-09-23:
> Is there a way to run CPython with only the standard library, or something similar, so it doesn't pick up packages like this?

How about a virtual environment?

dpgeorge, 2020-09-24:
It looks like `-IS` will work: it removes most things from sys.path, including site-packages.  It also has the advantage of approx doubling the speed of running the test suite because `site.py` is no longer run.

dpgeorge, 2020-09-24:
See #6478 for a proposed solution.  @stinos could you test that out?

stinos, 2020-09-24:
Works for me locally. Plus pretty nice that it's so much faster.

dpgeorge, 2020-09-25:
Fixed by 9123b67d641ba708a4ea3e5cd50665f0a73c6c8a

closed 2020-09-25 by dpgeorge