micropython/micropython #6487: no --recurse-submodule: 1st make errors, 2nd finishes.

CarlFK, 2020-09-26:
Looks like make checks out needed submodules, but ... not in time or something: 

```
sudo apt install gettext gcc-arm-none-eabi libnewlib-arm-none-eabi build-essential libffi-dev pkg-config
git clone https://github.com/micropython/micropython
cd micropython
cd mpy-cross
make
cd ../ports/stm32

make V=1 
stm32lib submodule not found, fetching it now...
(cd ../.. && git submodule update --init lib/stm32lib)
Submodule 'lib/stm32lib' (https://github.com/micropython/stm32lib) registered for path 'lib/stm32lib'
Cloning into '/home/juser/ev/micropython/lib/stm32lib'...
Submodule path 'lib/stm32lib': checked out '58fee7c92bd576814d3f2afd92fbc62990270ecc'
mkdir -p build-PYBV10/genhdr
GEN build-PYBV10/genhdr/pins.h
python3 boards/make-pins.py --board boards/PYBV10/pins.csv --af boards/stm32f405_af.csv \
	--prefix boards/stm32f4xx_prefix.c --hdr build-PYBV10/genhdr/pins.h --hdr-obj-decls \
	--qstr build-PYBV10/pins_qstr.h --af-const build-PYBV10/genhdr/pins_af_const.h \
	--af-defs build-PYBV10/genhdr/pins_af_defs.h --af-defs-cmp-strings \
	--af-py build-PYBV10/pins_af.py > build-PYBV10/pins_PYBV10.c
GEN stmconst build-PYBV10/modstm_qstr.h
python3 make-stmconst.py --qstr build-PYBV10/modstm_qstr.h --mpz build-PYBV10/genhdr/modstm_mpz.h ../../lib/stm32lib/CMSIS/STM32F4xx/Include/stm32f405xx.h > build-PYBV10/genhdr/modstm_const.h
python3 ../../py/makeversionhdr.py build-PYBV10/genhdr/mpversion.h
GEN build-PYBV10/genhdr/mpversion.h
GEN build-PYBV10/genhdr/moduledefs.h
python3 ../../py/makemoduledefs.py --vpath="., ../.., " lib/oofatfs/ff.c lib/oofatfs/ffunicode.c lib/littlefs/lfs2.c lib/littlefs/lfs2_util.c py/mpstate.c py/malloc.c py/gc.c py/pystack.c py/qstr.c py/vstr.c py/mpprint.c py/unicode.c py/mpz.c py/reader.c py/lexer.c py/parse.c py/scope.c py/compile.c py/emitcommon.c py/emitbc.c py/asmbase.c py/asmx64.c py/emitnx64.c py/asmx86.c py/emitnx86.c py/asmthumb.c py/emitnthumb.c py/emitinlinethumb.c py/asmarm.c py/emitnarm.c py/asmxtensa.c py/emitnxtensa.c py/emitinlinextensa.c py/emitnxtensawin.c py/formatfloat.c py/parsenumbase.c py/parsenum.c py/emitglue.c py/persistentcode.c py/runtime.c py/runtime_utils.c py/scheduler.c py/nativeglue.c py/pairheap.c py/ringbuf.c py/stackctrl.c py/argcheck.c py/warning.c py/profile.c py/map.c py/obj.c py/objarray.c py/objattrtuple.c py/objbool.c py/objboundmeth.c py/objcell.c py/objclosure.c py/objcomplex.c py/objdeque.c py/objdict.c py/objenumerate.c py/objexcept.c py/objfilter.c py/objfloat.c py/objfun.c py/objgenerator.c py/objgetitemiter.c py/objint.c py/objint_longlong.c py/objint_mpz.c py/objlist.c py/objmap.c py/objmodule.c py/objobject.c py/objpolyiter.c py/objproperty.c py/objnone.c py/objnamedtuple.c py/objrange.c py/objreversed.c py/objset.c py/objsingleton.c py/objslice.c py/objstr.c py/objstrunicode.c py/objstringio.c py/objtuple.c py/objtype.c py/objzip.c py/opmethods.c py/sequence.c py/stream.c py/binary.c py/builtinimport.c py/builtinevex.c py/builtinhelp.c py/modarray.c py/modbuiltins.c py/modcollections.c py/modgc.c py/modio.c py/modmath.c py/modcmath.c py/modmicropython.c py/modstruct.c py/modsys.c py/moduerrno.c py/modthread.c py/vm.c py/bc.c py/showbc.c py/repl.c py/smallint.c py/frozenmod.c extmod/moduasyncio.c extmod/moductypes.c extmod/modujson.c extmod/modure.c extmod/moduzlib.c extmod/moduheapq.c extmod/modutimeq.c extmod/moduhashlib.c extmod/moducryptolib.c extmod/modubinascii.c extmod/virtpin.c extmod/machine_mem.c extmod/machine_pinbase.c extmod/machine_signal.c extmod/machine_pulse.c extmod/machine_i2c.c extmod/machine_spi.c extmod/modbluetooth.c extmod/modussl_axtls.c extmod/modussl_mbedtls.c extmod/modurandom.c extmod/moduselect.c extmod/moduwebsocket.c extmod/modwebrepl.c extmod/modframebuf.c extmod/vfs.c extmod/vfs_blockdev.c extmod/vfs_reader.c extmod/vfs_posix.c extmod/vfs_posix_file.c extmod/vfs_fat.c extmod/vfs_fat_diskio.c extmod/vfs_fat_file.c extmod/vfs_lfs.c extmod/utime_mphal.c extmod/uos_dupterm.c lib/embed/abort_.c lib/utils/printf.c build-PYBV10/genhdr/moduledefs.h main.c stm32_it.c usbd_conf.c usbd_desc.c usbd_cdc_interface.c usbd_hid_interface.c usbd_msc_interface.c mphalport.c mpthreadport.c irq.c pendsv.c systick.c softtimer.c powerctrl.c powerctrlboot.c rfcore.c pybthread.c factoryreset.c timer.c led.c pin.c pin_defs_stm32.c pin_named_pins.c bufhelper.c dma.c i2c.c pyb_i2c.c spi.c pyb_spi.c qspi.c uart.c can.c fdcan.c pyb_can.c usb.c wdt.c eth.c gccollect.c help.c machine_adc.c machine_i2c.c machine_spi.c machine_timer.c machine_uart.c modmachine.c modpyb.c modstm.c moduos.c modutime.c modusocket.c network_lan.c modnetwork.c extint.c usrsw.c rng.c rtc.c flash.c flashbdev.c spibdev.c storage.c sdcard.c sdram.c fatfs_port.c lcd.c accel.c servo.c dac.c adc.c  lib/oofatfs/ff.c lib/oofatfs/ffunicode.c lib/littlefs/lfs2.c lib/littlefs/lfs2_util.c lib/libc/string0.c lib/mp-readline/readline.c lib/netutils/netutils.c lib/netutils/trace.c lib/netutils/dhcpserver.c lib/timeutils/timeutils.c lib/utils/gchelper_native.c lib/utils/pyexec.c lib/utils/interrupt_char.c lib/utils/sys_stdio_mphal.c lib/utils/mpirq.c extmod/modonewire.c > build-PYBV10/genhdr/moduledefs.h

....

LINK build-PYBV10/firmware.elf
arm-none-eabi-ld -nostdlib -L boards -Tboards/stm32f405.ld -Tboards/common_ifs.ld -Map=build-PYBV10/firmware.map --cref --defsym=_estack_reserve=8 --gc-sections -o build-PYBV10/firmware.elf build-PYBV10/py/mpstate.o build-PYBV10/py/nlr.o build-PYBV10/py/nlrx86.o build-PYBV10/py/nlrx64.o build-PYBV10/py/nlrthumb.o build-PYBV10/py/nlrpowerpc.o build-PYBV10/py/nlrxtensa.o build-PYBV10/py/nlrsetjmp.o build-PYBV10/py/malloc.o build-PYBV10/py/gc.o build-PYBV10/py/pystack.o build-PYBV10/py/qstr.o build-PYBV10/py/vstr.o build-PYBV10/py/mpprint.o build-PYBV10/py/unicode.o build-PYBV10/py/mpz.o build-PYBV10/py/reader.o build-PYBV10/py/lexer.o build-PYBV10/py/parse.o build-PYBV10/py/scope.o build-PYBV10/py/compile.o build-PYBV10/py/emitcommon.o build-PYBV10/py/emitbc.o build-PYBV10/py/asmbase.o build-PYBV10/py/asmx64.o build-PYBV10/py/emitnx64.o build-PYBV10/py/asmx86.o build-PYBV10/py/emitnx86.o build-PYBV10/py/asmthumb.o build-PYBV10/py/emitnthumb.o build-PYBV10/py/emitinlinethumb.o build-PYBV10/py/asmarm.o build-PYBV10/py/emitnarm.o build-PYBV10/py/asmxtensa.o build-PYBV10/py/emitnxtensa.o build-PYBV10/py/emitinlinextensa.o build-PYBV10/py/emitnxtensawin.o build-PYBV10/py/formatfloat.o build-PYBV10/py/parsenumbase.o build-PYBV10/py/parsenum.o build-PYBV10/py/emitglue.o build-PYBV10/py/persistentcode.o build-PYBV10/py/runtime.o build-PYBV10/py/runtime_utils.o build-PYBV10/py/scheduler.o build-PYBV10/py/nativeglue.o build-PYBV10/py/pairheap.o build-PYBV10/py/ringbuf.o build-PYBV10/py/stackctrl.o build-PYBV10/py/argcheck.o build-PYBV10/py/warning.o build-PYBV10/py/profile.o build-PYBV10/py/map.o build-PYBV10/py/obj.o build-PYBV10/py/objarray.o build-PYBV10/py/objattrtuple.o build-PYBV10/py/objbool.o build-PYBV10/py/objboundmeth.o build-PYBV10/py/objcell.o build-PYBV10/py/objclosure.o build-PYBV10/py/objcomplex.o build-PYBV10/py/objdeque.o build-PYBV10/py/objdict.o build-PYBV10/py/objenumerate.o build-PYBV10/py/objexcept.o build-PYBV10/py/objfilter.o build-PYBV10/py/objfloat.o build-PYBV10/py/objfun.o build-PYBV10/py/objgenerator.o build-PYBV10/py/objgetitemiter.o build-PYBV10/py/objint.o build-PYBV10/py/objint_longlong.o build-PYBV10/py/objint_mpz.o build-PYBV10/py/objlist.o build-PYBV10/py/objmap.o build-PYBV10/py/objmodule.o build-PYBV10/py/objobject.o build-PYBV10/py/objpolyiter.o build-PYBV10/py/objproperty.o build-PYBV10/py/objnone.o build-PYBV10/py/objnamedtuple.o build-PYBV10/py/objrange.o build-PYBV10/py/objreversed.o build-PYBV10/py/objset.o build-PYBV10/py/objsingleton.o build-PYBV10/py/objslice.o build-PYBV10/py/objstr.o build-PYBV10/py/objstrunicode.o build-PYBV10/py/objstringio.o build-PYBV10/py/objtuple.o build-PYBV10/py/objtype.o build-PYBV10/py/objzip.o build-PYBV10/py/opmethods.o build-PYBV10/py/sequence.o build-PYBV10/py/stream.o build-PYBV10/py/binary.o build-PYBV10/py/builtinimport.o build-PYBV10/py/builtinevex.o build-PYBV10/py/builtinhelp.o build-PYBV10/py/modarray.o build-PYBV10/py/modbuiltins.o build-PYBV10/py/modcollections.o build-PYBV10/py/modgc.o build-PYBV10/py/modio.o build-PYBV10/py/modmath.o build-PYBV10/py/modcmath.o build-PYBV10/py/modmicropython.o build-PYBV10/py/modstruct.o build-PYBV10/py/modsys.o build-PYBV10/py/moduerrno.o build-PYBV10/py/modthread.o build-PYBV10/py/vm.o build-PYBV10/py/bc.o build-PYBV10/py/showbc.o build-PYBV10/py/repl.o build-PY [... cut ...] ild-PYBV10/lib/oofatfs/ffunicode.o build-PYBV10/lib/littlefs/lfs2.o build-PYBV10/lib/littlefs/lfs2_util.o build-PYBV10/pins_PYBV10.o  "/usr/lib/gcc/arm-none-eabi/7.3.1/thumb/v7e-m/fpv4-sp/hard/libgcc.a"

arm-none-eabi-ld: cannot find build-PYBV10/lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/gcc/startup_stm32f405xx.o: No such file or directory
```
running make again, it builds.

starting with git clone --recurse-submodules https://github.com/micropython/micropython make runs fine the first time.

dlech, 2020-09-26:
I have seen this same issue before too. I don't really understand why it happens, but I worked around it in another project like this:

```make
# ensure required git submodules checked out
ifeq ("$(wildcard $(PBTOP)/micropython/README.md)","")
$(info GIT cloning micropython submodule)
$(info $(shell cd $(PBTOP) && git submodule update --init micropython))
ifeq ("$(wildcard $(PBTOP)/micropython/README.md)","")
$(error failed)
endif
```

This causes the command to be run and completed when the makefile is loaded rather than as a build target.

CarlFK, 2020-09-26:
I would rather make did not pull down more files.
make clean should put things back.   currently it does not delete the new files.

If you don't have the needed files, you should not expect good results.

stinos, 2020-09-26:
That's a valid point. On the other hand, having `make clean` erase everything would be equally wrong and hinder the build process: if I manually cloned submodules, should `make clean` still remove them? Moreover because of how dependency/qstr tracking works it's often needed to use the clean target, if it would delete half of the source each time build times get doubled or tripled for no good reason. 

Anyway if I remember correctly the reason submodules get updated automatically is the vast amount of 'help cannot build MicroPython because XXX is missing' issues and if you don't know what submodules are it ends there. Hence the decision to make the build self-contained (again, IIRC). Reverting that is not really an option, imo, but adding a make target like `deepclean` which does the exact inverse of the `submodules` target would be nice.

dpgeorge, 2020-10-01:
> I would rather make did not pull down more files.

Yes, I agree.

> Anyway if I remember correctly the reason submodules get updated automatically is the vast amount of 'help cannot build MicroPython because XXX is missing' issues and if you don't know what [...]

I think when we first introduced submodules (initially for stm32lib and axtls) Makefile rules were added to aid in the transition to submodules.

But I'd be happy to remove these rules now, there are only two instances, one in unix for axtls, and one in stm32 for stm32lib.

There's now a general `make submodules` rule for all ports which is used for explicit submodule fetching, if you don't want to do a full recursive checkout.

CarlFK, 2020-10-01:
> I think when we first introduced submodules (initially for stm32lib and axtls) Makefile rules were added to aid in the transition to submodules.
>
> But I'd be happy to remove these rules now, there [...]

How about changing the make action to give the user what they need:

```
# For convenience, automatically fetch required submodules if they don't exist
$(TOP)/lib/stm32lib/README.md:
	$(ECHO) "stm32lib submodule not found, can be fetched  with:"
	$(ECHO) "cd $(TOP) && git submodule update --init lib/stm32lib"
```

https://github.com/micropython/micropython/blob/master/ports/stm32/Makefile#L538

CarlFK, 2025-10-04:
I just tried again, following the docs, works fine.   so yeah, close it.

closed 2025-10-04 by dpgeorge