micropython/micropython #6489: ESP32 bluetooth.BLE.irq - TypeError: function doesn't take keyword arguments

jp-96, 2020-09-26:
Can't using keyword arguments.

For example, running ble_temperature.py in the bluetooth implementation example will raise a TypeError.

```
>>> from ble_temperature import demo
>>> demo()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "ble_temperature.py", line 78, in demo
  File "ble_temperature.py", line 38, in __init__
TypeError: function doesn't take keyword arguments
>>>
```

If deleting the keyword for the `BLE.irq()` argument in `__init__()`, it will be successfully executed.

``` python
class BLETemperature:
    def __init__(self, ble, name="mpy-temp"):
        self._ble = ble
        self._ble.active(True)
        #self._ble.irq(handler=self._irq)
        self._ble.irq(self._irq)
        ...
```

jimmo, 2020-09-26:
Thanks. We recently changed "handler" to no longer be a kwarg, but I forgot to update the examples.

I included a fix in a different PR here https://github.com/micropython/micropython/pull/6483/commits/abc3a5c067bf390b21017ce35ab6a9490cfd754a but should have split this out into its own PR to get it merged sooner. Will do that now.

closed 2020-09-26 by jp-96

FalconBFR, 2021-04-03:
Same issue here. Thanks and solved.
`#self._ble.irq(handler=self._irq)
        self._ble.irq(self._irq)`

PR #6490 (2020-09-26): examples/bluetooth: Update to use positional-only args to irq().
To match 6a6a5f9e151473bdcc1d14725d680691ff665a82. Docs were already updated.

Fixes #6489 - thanks @jp-96 sorry about the inconvenience!