micropython/micropython #6493: ESP32: dir(..) on some builtins causes code dump

robert-hh, 2020-09-27:
Calling dir(...) on some of the builtins cause a core dump. Tested tith no claim on completeness with the following items: 
zip, float, map, range, staticmethod, super, type, StopIteration, Stop AsyncIteration, complex, enumerate, filter, memoryview, reversed, slice, SystemExit, GeneratorExit, all Execption names like TypeError, NameError, ...
ESP8266 fails too.
Pyboard D behaves well, w600 port is fine.

Board Wems ESP32 Lite
MicroPython v1.13-77-g0fd0eb00a on 2020-09-27; ESP32 module with ESP32
Type "help()" for more information.

Core dump:
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e9096  PS      : 0x00060330  A0      : 0x800df59d  A1      : 0x3ffce3c0  
A2      : 0x3f40a018  A3      : 0x00000002  A4      : 0x3ffce4a0  A5      : 0x00000000  
A6      : 0x000000b3  A7      : 0x3ffce5f0  A8      : 0x00000000  A9      : 0x3ffce3c0  
A10     : 0x00000000  A11     : 0x3ffba2b0  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x3f40a018  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400e9096:0x3ffce3c0 0x400df59a:0x3ffce400 0x400df60c:0x3ffce420 0x400eaf31:0x3ffce4a0 0x400e394e:0x3ffce4d0 0x400df881:0x3ffce500 0x400ed3ad:0x3ffce520 0x400e3a08:0x3ffce5c0 0x400df881:0x3ffce610 0x400df8aa:0x3ffce630 0x40104964:0x3ffce650 0x40104b76:0x3ffce6e0 0x400f62fc:0x3ffce720 0x4009b54a:0x3ffce750

For some reason I could not decode the backtrace

jimmo, 2020-09-27:
Possibly related to #6306 ? (Haven't tested this myself yet)

robert-hh, 2020-09-27:
Adding that change indeed  fixes it.

mikewadsten, 2020-09-27:
> Adding that change indeed fixes it. The question then is: why does it not tip over on the STM32 port.

I'm not 100% sure but I know on other microcontrollers without an MMU (or without using the MMU), things like accessing a null pointer or "writing" to a flash address or bad pointer does not necessarily trigger a hard fault.

robert-hh, 2020-09-27:
@mikewadsten You may be right, because in that case only read accesses are made.

Josverl, 2020-10-03:
@robert-hh 
I see issues on the PYBv1.1 as well with the 1.13 build:
MicroPython v1.13 on 2020-09-02; PYBv1.1 with STM32F405RG
```
import io
dir(io.IOBase)
# Expected : 
# ['__class__', '__name__', '__bases__']'
# Actual: 
# MemoryError: memory allocation failed, allocating 1075984344 bytes
```
while it does not crash the CPU, the memory allocation error does not look good either

robert-hh, 2020-10-04:
Obviously it is a bug, a zero pointer de-reference, which results in different behavior depending on the CPU type.

Josverl, 2020-10-07:
Agree , just pointed out that it is able to repro on STM32 as well, 
which may make it simpler to add to a test suite without the need to add too much different hardware models>

did not try it on unix though.

dpgeorge, 2020-10-09:
Thanks for the reports.  Should be fixed by b137d064e9e0bfebd2a59a9b312935031252e742

closed 2020-10-09 by dpgeorge

Josverl, 2020-10-10:
confirm fixed 
, i tested this on esp32 by running the stubber script that tries to enum just about everything and it runs fine.