micropython/micropython #6495: Sparkfun ESP32 "Things" - DNS Not "Sticking"?

jwshields, 2020-09-28:
Hello all,
This is my first issue in this repo, please forgive if I leave out any information. I am also relatively new to microcontrollers/micropython/SoC boards, and the like; still learning the layout of the land..

I'm experiencing an odd issue on two of my ESP32 boards, and I am unsure what this is stemming from or if this is due to something I am doing improperly.

Problem: When I connect my board to my wireless network, it will connect properly and `nic.ifconfig()` will report proper parameters for some time, but then, while still remaining connected to the network, **my DNS settings in `nic.ifconfig()` will change unprompted.**
I am sure I do not have a rogue DHCP server on my network, nor anything that is advertising/hijacking DNS traffic.

My configuration:
Firmware: esp32-idf3-20200902-v1.13.bin
Boards: [Sparkfun ESP32 Thing](https://www.sparkfun.com/products/13907) & [SparkFun Thing Plus - ESP32 WROOM](https://www.sparkfun.com/products/15663)
Router: Mikrotik RB4011iGS+5HacQ2HnD (DHCP Server, configured to advertise Pi-Hole for DNS)
DNS: Pi-Hole v5.1.2

Reproduction:
- Flash board per instructions on Micropython website for ESP32 [(Link)](https://micropython.org/download/esp32/)
- Connect to board for REPL via preferred serial client
- Python:
```
>>> import network
>>> nic = network.WLAN(network.STA_IF)
>>> nic.active(True)
>>> nic.config(dhcp_hostname="esp32_thing")
>>> nic.connect('myssid', 'secret')
>>> nic.ifconfig()
```
- I should receive output similar to this, and generally do right after connecting:
```>>> nic.ifconfig()
('192.168.89.235', '255.255.254.0', '192.168.88.1', '192.168.88.9')
```

- Some indeterminate amount of time later, if I execute `nic.ifconfig()` again, I receive output that has the DNS server changed.

Example code,
```
def connect_and_print():
         nic.disconnect()
         time.sleep(2)
         nic.connect("myssid", "secret")
         while not nic.isconnected():
             time.sleep_ms(250)
             print("Connecting...")
         print("Connected")
         dns_expect = "192.168.88.9"
         print(dns_expect)
         print(nic.ifconfig())
         while nic.ifconfig()[3] == dns_expect:
             print("DNS OK")
             time.sleep_ms(750)
         print("DNS CHANGED!")
         print(nic.ifconfig())
connect_and_print()
```

Earlier I used "indeterminate" because I have not seen any consistency with when or how this happens.
It could be within 10 seconds of connecting to the network, or 5 minutes.
I expect my DNS settings to remain "192.168.88.9" - but often they change to an IP something like "253.222.x.x"

Has anybody got an idea of what's happening here? While this is not hugely inconveniencing to me, it is concerning. My plans don't intend for these boards to be reaching out to the internet often, if at all, but also, I would prefer my local DNS server settings to be retained, in case they need to resolve a local address..

UnexpectedMaker, 2020-09-28:
Why are you using IDF3? There are 1001 fixes in IDF4, so maybe try ESP32 firmware built from that and see if it exhibits the same issues?

jwshields, 2020-09-28:
The page which provides the firmware downloads for Micropython suggests this: `Firmware is provided using either ESP-IDF v3.x or v4.x. If in doubt use v3.x.`
I will be giving 4.x a try now

jwshields, 2020-09-28:
I have tested `esp32-idf4-20200902-v1.13.bin` on both the Sparkfun ESP32 Thing, and Sparkfun ESP32 Thing - WROOM. In the few hours since flashing, I have seen neither of the boards have changes to DNS.

I will continue to run idf4 on my boards, but this leaves me with a few questions.
1. Why does the documentation for the ESP32 (https://micropython.org/download/esp32/) - suggest using idf3?
2. What is going on in Micropython that the idf3 version that is causing this weirdness, that is not present in idf4?
3. Is there a resource I can look at which shows the differences in the idf toolsets? (unrelated to micropython)

I guess this issue could be closed, but as I mentioned in the original post, I am fairly new to all of this, and the suggestion on the download page of using the idf3 firmware seems to be misleading.

jonnor, 2024-09-13:
We are now on esp-idf 5.x. Bugs get fixed in new esp-idf versions all the time (and sometimes new ones are introduced).

closed 2024-09-13 by jonnor