micropython/micropython #6497: Pinout use for nrf52840 dongle

JPFrancoia, 2020-09-29:
Hello,

I'm trying to understand how the pins of the nrf52840 are mapped/numbered in the board definition.

Looking at the `pins.csv` file, I see pins going as high as P47:

https://github.com/micropython/micropython/blob/0fd0eb00aa5b9d311046d48d73a8cfabb30d7dd6/ports/nrf/boards/pca10059/pins.csv#L28

But there isn't such a pin on the board. Here is the pinout of the board (from the doc):

[image]

I wasn't able to find out what numbering scheme was used to number the pins. It makes it hard to know to which physical pin a  pin  `Pin` object is tied (in Micropython).

Could someone help me on this please? Maybe I'm missing something. @glennrub maybe (I think you built the file)?

jimmo, 2020-10-01:
@JPFrancoia 
0-31 correspond to 0.x and 32-63 correspond to 1.x

So 47 would be 1.15 on the PCA10059

JPFrancoia, 2020-10-01:
Great, thank you! I opened a PR to document this: https://github.com/micropython/micropython/pull/6502