micropython/micropython #6498: Concurrent BLE advertising/scanning on STM32WB

Hendrik-K, 2020-09-30:
Hi,

I encountered an unexpected behaviour while using BLE on my STM32WB (NUCLEO-WB55).

When trying to scan while advertising, an OSError is raised.

I tested the behaviour by running a little code snippet:

```
import bluetooth
ble = bluetooth.BLE()
ble.active(True)
def irq(event, data):
    print("event: {} data: {}".format(event, data))
adv_data = b'\x05\x09\x54\x45\x53\x54'
ble.gap_advertise(50000, adv_data, connectable=False)
ble.gap_scan(1000)
Traceback (most recent call last):                                              
  File "<stdin>", line 1, in <module>                                          
OSError: [Errno 5] EIO
```

I originally expected the implementation to support concurrent advertising/scanning as the hardware does.
It would be nice if someone could explain if this behaviour is as it should be or if this is an error.

Thanks!

dpgeorge, 2020-09-30:
Try without the `connectable=False`, and also register the irq handler, eg:
```python
import bluetooth
ble = bluetooth.BLE()
ble.active(True)
def irq(event, data):
    print("event: {} data: {}".format(event, data))
ble.irq(irq)
adv_data = b'\x05\x09\x54\x45\x53\x54'
ble.gap_advertise(50000, adv_data)
ble.gap_scan(5000)
```

Hendrik-K, 2020-09-30:
Thanks for the reply, but unfortunately this has the same outcome for me as earlier.

dpgeorge, 2020-10-01:
The above code works on a PYBD-SF2, which uses the same BLE stack (NimBLE) as the WB55.  So it looks like a limitation of the WB55 itself.

You may need to upgrade the wireless stack on the WB55.  To determine the current version you can run the `ports/stm32/boards/NUCLEO_WB55/rfcore.py` script on the device.

Hendrik-K, 2020-10-01:
Thanks for the clarification.
I had already installed the latest HCI layer wireless stack (1.8.0), but I ran the script anyway.
```wireless stack version   : 0x01080004 (1.8.0.0.4) ```

That's strange that the WB55 does not behave like expected.

In [this](https://www.st.com/resource/en/programming_manual/dm00716581-stm32wb-ble-stack-programming-guidelines-stmicroelectronics.pdf) document from ST it says on page 4:

>... 
STM32WB Current features supported on v4.1:
– Multiple roles simultaneously support
**– Support simultaneous advertising and scanning**
...

dpgeorge, 2020-10-09:
I'm not sure the WB55 yet supports simultaneous scanning and advertising.  In the very latest WS update V1.9.0 / 21-September-2020, there is a new WS firmware:
- stm32wb5x_BLE_HCI_AdvScan_fw.bin : Ability to do advertising and scanning through HCI interface,
Any connection can’t be done, numOfLinks can be set to 1 at the stack initialization call

It looks like that WS is observer/broadcaster role only, ie cannot do connections.  I did quickly try it but it still gave the same EIO error when doing both scanning and advertising (and the scanning didn't seem to work at all...).

So for the time being I'd say that simultaneous scanning and advertising is not supported on the WB55, and it's a limitation of the WS firmware (the hardware should be able to do it, and NimBLE can do it).

Hendrik-K, 2020-10-09:
Thank you very much for your thorough investigation of this matter.
It's good to know that the limitation is on the WS firmware side.

FathiMahdi, 2023-10-31:
@Hendrik-K from where you got the Bluetooth API  ?

Sikandar777, 2025-03-04:
have you try to generate the new file with cubemx?