micropython/micropython #6499: Version info reported on boot is incorrect

peterhinch, 2020-09-30:
Current daily build as listed on website:
PYBD-SF2-20200930-unstable-v1.13-77-g0fd0eb00a.dfu (latest)

Report on boot using firmware built from today's sources.
Unix:
MicroPython v1.12-778-g0fd0eb00a-dirty on 2020-09-30; linux version
Pyboard D SF6W:
MicroPython v1.12-778-g0fd0eb00a-dirty on 2020-09-30; PYBD-SF6W with STM32F767IIK

Pyboard D SF6W REPL:
```python
>>> os.uname()
(sysname='pyboard', nodename='pyboard', release='1.13.0', version='v1.12-778-g0fd0eb00a-dirty on 2020-09-30', machine='PYBD-SF6W with STM32F767IIK')
```

dpgeorge, 2020-09-30:
Did you `git pull --tags`?  Making a fresh clone of this repo, I get:

peterhinch, 2020-10-01:
Thank you - that fixed it I've updated my build script accordingly.

closed 2020-10-01 by peterhinch