micropython/micropython #6506: ESP32 SDMMC with pin reassignment

d51r3verse, 2020-10-02:
I prefer SD_MMC(3 signal wires) than SPI methods(4 signal wires).
But, It seems to be fixed on ESP-IEF SDK [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sdmmc_host.html#overview) and, [here](https://github.com/espressif/esp-idf/blob/f676a3b1906f3bcec61f870009f4f3d840d38e58/components/driver/include/driver/sdmmc_host.h#L82)
```
uSD | 1-bit Slot1 | SPI(native2,3/GPIO)
----+-------------+---------------------
 3  |   CMD(15)   | MOSI(13,23/any)
 5  |   CLK(14)   | SCK(14,18/any)
 7  |   D0(2)     | MISO(12,19/any)
```
I wish to avoid using strapping pins. I know workaround for GPIO12(burn efuse for VDD regs) only.
But I don't know how to workaround GPIO2 while I using SDMMC mode.
Any suggestions?

pumelo, 2020-10-17:
Are you talking about the pull-ups required to make emmc work and which will make entering boot-mode impossible?
the easy solution to this is to use the internal pull-up rather then external. After boot you enable the pull-up by software. the issue is that the internal pull-ups are weak.. which might not work in your application ...

Other than that you can use custom circuitry to pull GPIO2 low when you like to enter boot.

jonnor, 2025-10-03:
Is this still an issue?