micropython/micropython #6509: Crash on tab-completion of module.__class__ properties

jdstroy, 2020-10-03:
Transcript from an interactive session:

```
>>> import sys
>>> sys.__class__
<class 'module'>
>>> sys.__class__.Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e8eea  PS      : 0x00060630  A0      : 0x800df489  A1      : 0x3ffceb20
A2      : 0x00000002  A3      : 0x000000b3  A4      : 0x3ffcec00  A5      : 0x00000000
A6      : 0x000000b3  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffceb20
A10     : 0x00000000  A11     : 0x3ffba28c  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x3f408b90  A15     : 0x00001a3a  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x400e8ee7:0x3ffceb20 0x400df486:0x3ffceb60 0x400df4f8:0x3ffceb80 0x400edbdd:0x3ffcec00 0x40103c99:0x3ffcec50 0x40103ec8:0x3ffcec80 0x401044e3:0x3ffceca0 0x400f5cdc:0x3ffcece0 0x4009b981:0x3ffced10

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5148
load:0x40078000,len:12880
load:0x40080400,len:3484
entry 0x40080630
```

Micropython on ESP32 (`esp32-idf4-20200902-v1.13.bin`) crashes when trying to tab-complete properties of module objects.

jdstroy, 2020-10-03:
Similar issue with tab completion on type objects:

```
>>>
>>> x = 0
>>> x.__class__.__class__
<class 'type'>
>>> x.__class__.__class__.Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e8eea  PS      : 0x00060630  A0      : 0x800df489  A1      : 0x3ffceb20
A2      : 0x00000002  A3      : 0x000000b3  A4      : 0x3ffcec00  A5      : 0x00000000
A6      : 0x000000b3  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffceb20
A10     : 0x00000000  A11     : 0x3ffba28c  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x3f409ed4  A15     : 0x000027aa  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x400e8ee7:0x3ffceb20 0x400df486:0x3ffceb60 0x400df4f8:0x3ffceb80 0x400edbdd:0x3ffcec00 0x40103c99:0x3ffcec50 0x40103ec8:0x3ffcec80 0x401044e3:0x3ffceca0 0x400f5cdc:0x3ffcece0 0x4009b981:0x3ffced10

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5148
load:0x40078000,len:12880
load:0x40080400,len:3484
entry 0x40080630
```

robert-hh, 2020-10-03:
Similar to  https://github.com/micropython/micropython/pull/6306 and https://github.com/micropython/micropython/issues/6493.
The change suggested in #6306 fixes it.

dpgeorge, 2020-10-09:
Should be fixed by b137d064e9e0bfebd2a59a9b312935031252e742

closed 2020-10-09 by dpgeorge