micropython/micropython #6517: pyb Pyboard D SD card detection is broken

peterhinch, 2020-10-06:
The [official docs](https://pybd.io/hw/pybd_sfxw.html#sd-card) specify using this (which used to work):
```python
import sys, os, pyb

if pyb.SDCard().present():
    os.mount(pyb.SDCard(), '/sd')
    sys.path[1:1] = ['/sd', '/sd/lib']
```
I'm seeing
```python
MicroPython v1.13-94-g817b80a10-dirty on 2020-10-02; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> 
>>> import pyb
>>> pyb.SDCard.present()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 1 positional arguments but 0 were given
```
What argument does it now require?

dpgeorge, 2020-10-06:
> ```python
> pyb.SDCard.present()
> ```

You need `pyb.SDCard().present()`.

peterhinch, 2020-10-06:
Aagh. Sorry. Struggling with eyesight problems today :-(

closed 2020-10-06 by peterhinch