micropython/micropython #6518: uasycio Loop close()

kr-g, 2020-10-06:
this belongs to an issue reported in following micropython forum thread.

https://forum.micropython.org/viewtopic.php?f=15&t=9103

the asyncio Loop close() method currently does not perform any action.
whereas the new_event_loop() method implements the functionality.

in CPython its documented as 
https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.loop.close

performing the same functionality in close() would resolve this behaviour, 
and would make uasyncio more compatible to CPython.

dpgeorge, 2020-10-06:
> the asyncio Loop close() method currently does not perform any action.
> whereas the new_event_loop() method implements the functionality.
> ...
> performing the same functionality in close() would resolve this behaviour,
> and would make uasyncio more compatible to CPython.

Can you provide some sample code that runs under CPython and MicroPython and shows the discrepancy between the two of them?  From what I understand, once you call `loop.close()` you cannot retrieve the event loop again and reuse it, and instead must call `new_event_loop()` (so it doesn't matter whether the logic is in close or new_event_loop).

kr-g, 2020-10-06:
precisely.
thats also what i understood from the problem raised by jpbihin.
it's not regarding reusing the loop. it's regarding shuting down in a well defined manner where the logic is more like calling close() rather then new_event_loop() what is quite hard to understand whilst reading.
it is regarding that close() in micropython does currently nothing expect pass statement (as shown by code reference provided by pythoncoder), and this is clashing (as stated out in documentation) with CPython where it do implement similar logic as new_event_loop.
this report adresses this discrepancy for further discussion when to resolve this mismatch.
clearly that  new_event_loop() needs to be called in conjunction with set_event_loop(), or any of the other method which create also internally a new event loop if none is given (refer to https://docs.python.org/3/library/asyncio-eventloop.html) in case processing should continue afterwards.

kr-g, 2020-10-06:
copy of jpbihin's code from the forum post. (see also link to post above)

import asyncio

import time
import sys

async def fn_1() :
    while True :
        print("fn_1") 
        print(time.time())
        await asyncio.sleep(4)

loop = asyncio.get_event_loop()
loop.create_task(fn_1())

try :
    loop.run_forever()

except KeyboardInterrupt :
    print("keyboard interrupt")
    sys.exit()
    
except Exception as e:
    print(e)
    sys.exit()
    
finally:
    loop.close()

dpgeorge, 2020-10-06:
> it's regarding shuting down in a well defined manner where the logic is more like calling close()

There is no shutting down to do, that's why `close()` does nothing.  At best it could just set to `None` the global queue variables.

> copy of jpbihin's code from the forum post

From the forum post, as I understand it, the desire is to run the code again after breaking it with ctrl-C.  But that won't work on CPython either because you cannot use the loop after `loop.close()` is called.

kevinkk525, 2020-10-07:
I think setting the queue to None could be beneficial/cleaner as the resources can then be garbage collected.
However, starting a new loop in micropython comes with some difficulties due to its micro implementation, e.g. the main queue might be empty but you have to manually remove/recreate all objects that retain their own waiting queues like Events and Locks. All tasks waiting for an Event are being stored in the Event queue unil the event is set. There is no other reference to these tasks. So stopping the main loop and creating a new one doesn't touch those tasks. Therefore the next time you set that Event, all those old tasks will be put back on the new loop. That won't happen with CPython. I'd say using different loops in micropython is rather dangerous and unsupported/not advised.

kr-g, 2020-10-07:
in fact its complicated. 
just prepared sort of playground.

```
_micropython = True
try:
    import uasyncio as asyncio
except:
    import asyncio
    _micropython = False

import time
import sys

async def fn_1() :
    cnt = 0
    while True :
        print("fn_1") 
        print(time.time())
        await asyncio.sleep(4)
        cnt += 1
        if cnt > 3:
            # this just simulates the user interaction
            # comment out to test other exception
            raise KeyboardInterrupt()
            # this simulates an error case
            raise Exception("something went wrong")

# use this bool switch for test 2 scenarios
run_with_loop_close_and_create_new_loop = False

# CPython: this creates an event loop in case none is existing for the current thread
loop = asyncio.get_event_loop()
loop.create_task(fn_1())

try :
    loop.run_forever()

except KeyboardInterrupt :
    print("keyboard interrupt")
    
except Exception as e:
    print("loop error",e)    

finally:
    if run_with_loop_close_and_create_new_loop:
        loop.close()
 

if run_with_loop_close_and_create_new_loop:
    print("setting up loop for current task after close was called")
    if _micropython:
        asyncio.new_event_loop()
    else:
        asyncio.set_event_loop( asyncio.new_event_loop() )

print("doing again")

loop = asyncio.get_event_loop()
loop.create_task(fn_1())

try :
    loop.run_forever()

except KeyboardInterrupt :
    print("keyboard interrupt")
    
except Exception as e:
    print("loop error",e)    

finally:
    if run_with_loop_close_and_create_new_loop:
        loop.close()

 
print("done")
sys.exit() 
```

kr-g, 2020-10-07:
one remark:
when setting ```run_with_loop_close_and_create_new_loop = False```
tasks are not queued up again (yesterday this was the case on my esp32)

peterhinch, 2020-10-07:
This is academic for most firmware apps which start a single event loop and run forever. @kevinkk525 I agree with your comment deprecating the use of multiple loops. I can't recall any firmware application which did this.

Yet I find **new_event_loop** a great improvement over V2 when running test scripts, letting you run further scripts without a reboot. @kevinkk525 I take your point about Events and Locks, I guess it would be good if there were a way of clearing their stored state. I've never hit that problem because my scripts complete with no tasks waiting on these objects. Clearly if you ctrl-c that might not be the case.

kr-g, 2020-10-07:
i agree that the sample quite theoretical.
but catching an exception and continue processing is a real world problem.

beside that calling ```close()``` is more intuitive than calling ```new_event_loop()```

kevinkk525, 2020-10-07:
continuing is a real world problem but you don't continue by creating a new loop. In all my projects creating a new loop would do nothing because you have to create the tasks again, so a hard/soft-reset is the only way of starting over. Continuing after issuing ctrl+c can easily be done by making the loop.run_forever() again.
Besides, the only exceptions you can catch from loop.run_forever() are the ctrl+c interrupt and bugs in uasyncio itself. All uasyncio tasks handle exceptions differently.

kr-g, 2020-10-07:
> Besides, the only exceptions you can catch from loop.run_forever() are the ctrl+c interrupt and bugs in uasyncio itself. All uasyncio tasks handle exceptions differently.

absolutely correct

> in all my projects creating a new loop would do nothing because you have to create the tasks again,

you are not alone (...also a song by michael jackson)
there might be plenty of scenarios on how to deal with it

t35tB0t, 2020-10-29:
FWIW:  I ran into somewhat similar issues.  CTRL-C isn't an interrupt but is set an exception.  I wanted to cleanly halt asyncio tasks so I could inspect variables, execute some functions, then resume cleanly.  What's worked here is added a SIGQUIT trap in the USB CDC handler to set a reset-on-read bool flag in the micropython VM.  Then, in an async supervisory task, we poll the SIGQUIT flag to issue a get_event_loop().stop()   

After poking around with REPL, then can resume with asyncio.get_event_loop().run_forever()

If there's anything bad about doing this - it sure hasn't bit us yet.  The problem with CTRL-C keyboard Exception is that it it does not create a controlled "re-entrant" halt.  The new trap does.  This became part of an cross-platform asyncio application/task framework with RESTful APIs.  So the above trap launches with equivalent mappings for Windoz SIGBREAK and Linux SIGQUIT.  So same benign halt/resume behavior experience on all three platforms.  

A takeaway here is that it might be useful to officially have a few system signals supported ala Linux cPython via the signal module (SIGQUIT being the most obvious here).

kr-g, 2020-10-29:
interesting. which framework?

t35tB0t, 2020-10-30:
Framework?  it is just a t35tB0t-crafted framework to tame the system (assuming its even worthy of such descriptions).  What matter here is that the new  asyncio V3 is awesome.   Our micro-framework is just a neat "ribbon" of code around that.   Basically, we've crafted a stable and robust system for dynamic loading of tasks and webapps around asyncio.

We modified and hardened both PicoWeb and uaioftp; adding a directory structure and built a async task manager along with a web apps manager (as a task).  At boot, the system dynamically finds, registers, and runs all the task-class based modules found in the task folder.  The web apps mgr does the same (picoweb-route mount style) with all the web apps found in the web apps folder.  The core boots, handles networking, syncs up with time server, and "just works".

The aioftp server has been substantially extended to support multiple sessions, and more useful FTP commands such as retries etc.  We tweaked up the stack settings to permit many more connections with more optimized (smaller) packets.  This makes it easy to run eight concurrent FTP file transfers and still server up simple web services while leaving the majority of system ram free on a STM32F767.  The "framework" aka aiocore itself is pretty flat, reasonably small, and the usage is reasily scalable due to dynamic loading of modules.  Bulk transfer rates rank up around 7.5Mbps when transferring from the SDcard.

kr-g, 2020-10-30:
i meant mainly to get an url to this interesting project (in case its open source)

anyway - its off topic from the issue reported here.

probably you can provide more information in the forum regarding this ?

the initial request came from the forum board port here
https://forum.micropython.org/viewtopic.php?f=15&t=9103