micropython/micropython #6519: ESP32 doesn't build on latest master

elliotwoods, 2020-10-06:
I'm getting many errors in the `berkley-db-1.xx` folder.
Some of the errors from `make -j` here:

https://gist.github.com/elliotwoods/e0d266105b23054edef8f6bcce0067de/edit

dpgeorge, 2020-10-06:
It builds locally for me, and also on Travis CI.  So it's likely something to do with your build environment.  To debug further, please specify your exact set-up and full steps to reproduce the error.

elliotwoods, 2020-10-06:
Coincidentally I have exactly the notes on how to recreate this as I've been writing build instructions as I've been going along for the project I'm using this for:

https://gist.github.com/elliotwoods/59e7dd58c53e1ee8f54cbb4437ec222f

dpgeorge, 2020-10-06:
Using Windows, there may be problems with symlinks in `berkley-db-1.xx`.  Try disabling that via `make MICROPY_PY_BTREE=0`

elliotwoods, 2020-10-06:
good call. testing now
although it is on linux, it's on an ntfs file system

elliotwoods, 2020-10-06:
yep! that's the one!

builds succesfully now

I was wondering why db.h just contains a path

elliotwoods, 2020-10-06:
Changing the contents of each symlink (that i found) in the `PORT/include` folder of berkley db from the path to `#include "path"` fixes the build.

Changes:
```
diff --git a/PORT/include/bsd-queue.h b/PORT/include/bsd-queue.h
index f28ad5d..462f7c0 120000
--- a/PORT/include/bsd-queue.h
+++ b/PORT/include/bsd-queue.h
@@ -1 +1 @@
-queue.h
\ No newline at end of file
+#include "queue.h"
\ No newline at end of file
diff --git a/PORT/include/db.h b/PORT/include/db.h
index 44c1ba4..a3c0e4e 120000
--- a/PORT/include/db.h
+++ b/PORT/include/db.h
@@ -1 +1 @@
-../../include/db.h
\ No newline at end of file
+#include "../../include/db.h"
\ No newline at end of file
diff --git a/PORT/include/filevtable.h b/PORT/include/filevtable.h
index 25306d1..7213be5 120000
--- a/PORT/include/filevtable.h
+++ b/PORT/include/filevtable.h
@@ -1 +1 @@
-../../include/filevtable.h
\ No newline at end of file
+#include "../../include/filevtable.h"
\ No newline at end of file
diff --git a/PORT/include/mpool.h b/PORT/include/mpool.h
index 03f870c..4fdd0df 120000
--- a/PORT/include/mpool.h
+++ b/PORT/include/mpool.h
@@ -1 +1 @@
-../../include/mpool.h
\ No newline at end of file
+#include "../../include/mpool.h"
\ No newline at end of file
```

dpgeorge, 2020-10-06:
Ok, good to know symlinks are the issue.

This is a mildly common problem.  Solution would be to update the submodules that use symlinks to not use symlinks (berkley and axtls).

stinos, 2020-10-07:
NTFS pretty much always supported symlinks and recent git versions definitely can handle it on Windows: with `core.symlinks=true` I can just clone MicroPython and submodules and files like db.h are turned into proper symlinks:
```
 l .\micropython\lib\berkeley-db-1.xx\PORT\include

Mode                LastWriteTime     Length Name
----                -------------     ------ ----
-a---l       07/10/2020     09:02        0   bsd-queue.h
-a----       07/10/2020     09:02     4.58KB cdefs.h
-a----       07/10/2020     09:02     7.13KB compat.h
-a---l       07/10/2020     09:02        0   db.h
-a---l       07/10/2020     09:02        0   filevtable.h
-a---l       07/10/2020     09:02        0   mpool.h
-a----       07/10/2020     09:02     2.75KB ndbm.h
-a----       07/10/2020     09:02     8.41KB queue.h
```
Same on WSL:
```
l ./micropython/lib/berkeley-db-1.xx/PORT/include/
total 28K
drwxr-xr-x 1 stijn stijn 4.0K Oct  7 09:12 ./
drwxr-xr-x 1 stijn stijn 4.0K Oct  7 09:12 ../
lrwxrwxrwx 1 stijn stijn    7 Oct  7 09:12 bsd-queue.h -> queue.h
-rw-r--r-- 1 stijn stijn 4.6K Oct  7 09:12 cdefs.h
-rw-r--r-- 1 stijn stijn 7.2K Oct  7 09:12 compat.h
lrwxrwxrwx 1 stijn stijn   18 Oct  7 09:12 db.h -> ../../include/db.h
lrwxrwxrwx 1 stijn stijn   26 Oct  7 09:12 filevtable.h -> ../../include/filevtable.h
lrwxrwxrwx 1 stijn stijn   21 Oct  7 09:12 mpool.h -> ../../include/mpool.h
-rw-r--r-- 1 stijn stijn 2.8K Oct  7 09:12 ndbm.h
-rw-r--r-- 1 stijn stijn 8.5K Oct  7 09:12 queue.h
```

So I think a better solution is to make this clear in the README (people should just have a recent git version and do `git config --global core.symlinks true`, there's really no compelling reason to not use it) and/or change the Makefile's submodule target to clone with e.g. `git clone -c core.symlinks=true`.

mattytrentini, 2020-10-23:
Do you know if `core.symlinks` can be configured to operate correctly when running Docker containers on Windows hosts @stinos? That's how I often run into this error; using a volume mount to a MicroPython clone in Windows _inside the container_ has always presented a problem...

stinos, 2020-10-23:
No experience with that, but from your explanation it sounds as if core.symlinks cannot do anything there: it only affects cloning, if you already have cloned and whatever environment you use doesn't recognize the symlinks on the volume (or you initially cloned without symlinks) it's too late. But e.g. WSL, MSys2, Cygwin all handle Windows symlinks correctly in my experience, can you elaborate on how you use Docker containers? Is a workaround like cloning from within the container doable?

elliotwoods, 2020-10-23:
It might be worth noting that in my case I likely:

1. Cloned in git bash for windows (which is much quicker than WSL for network and disk operations)
2. Ran make from WSL Ubuntu

stinos, 2020-10-23:
That's a case which should work fine though ('git bash' is MSYS2 afaik):
Start git-bash.exe
```
> cd /c/
> git clone git@github.com:micropython/micropython.git test
> git submodule update --init
```
Start WSL:
```
> l /mnt/c/test/lib/berkeley-db-1.xx/PORT/include
...
lrwxrwxrwx 1 stijn stijn    7 Oct 23 09:44 bsd-queue.h -> queue.h*
lrwxrwxrwx 1 stijn stijn   26 Oct 23 09:44 filevtable.h -> ../../include/filevtable.h*
lrwxrwxrwx 1 stijn stijn   21 Oct 23 09:44 mpool.h -> ../../include/mpool.h*
...
> make -j4 -C /mnt/c/test/mpy-cross
> ~/projects/esp32/export.sh
> make -j4 -C /mnt/c/test/ports/esp32
....
CC ../../lib/berkeley-db-1.xx/btree/bt_close.c
...
LINK build-GENERIC/application.elf
   text    data     bss     dec     hex filename
1129312  270536   37428 1437276  15ee5c build-GENERIC/application.elf
...
```

mattytrentini, 2020-10-23:
> No experience with that, but from your explanation it sounds as if core.symlinks cannot do anything there: it only affects cloning, if you already have cloned and whatever environment you use [...]

Clone a repo in Windows. Start a container, mounting the part of the filesystem containing the clone inside the container and then begin `make`'ing (stm32 currently). The problem arises because `make submodules` occurs _inside the container_ but the container doesn't know that the underlying filesystem is Windows-based so symlinks are not handled correctly.

A workaround ought to be to manually init submodules directly in Windows (with `core.symlinks` enabled) but that is a _little_ awkward...

> Is a workaround like cloning from within the container doable?

Possible but won't help - you'll still need a volume mount so that the files persist. Which means you'll _still_ have the same symlink issue.  

Might be time to set up WSL!

stinos, 2020-10-23:
> Start a container

My question was rather: how do you do this, which software do you use that doesn't handle symlinks?

markmelvin, 2020-12-18:
This kinda sucks. micropython used to work fine on WSL/WSL2 but I tried to build v1.13 today and ran into this issue. I have not been able to get it to build yet. The problem I am seeing is:

```
CC ../../extmod/modussl_axtls.c
In file included from ../../lib/axtls/ssl/tls1.h:46,
                 from ../../lib/axtls/ssl/ssl.h:75,
                 from ../../extmod/modussl_axtls.c:36:
../../lib/axtls/ssl/os_port.h:1:20: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘.’ token
    1 | os_port_micropython.h
      |                    ^
In file included from ../../lib/axtls/crypto/crypto.h:42,
                 from ../../lib/axtls/ssl/tls1.h:47,
                 from ../../lib/axtls/ssl/ssl.h:75,
                 from ../../extmod/modussl_axtls.c:36:
../../lib/axtls/crypto/bigint_impl.h:89:5: error: unknown type name ‘comp’
   89 |     comp* comps;                /**< A ptr to the actual component data */
      |     ^~~~
In file included from ../../lib/axtls/crypto/crypto.h:43,
                 from ../../lib/axtls/ssl/tls1.h:47,
                 from ../../lib/axtls/ssl/ssl.h:75,
                 from ../../extmod/modussl_axtls.c:36:
../../lib/axtls/crypto/bigint.h:46:32: error: unknown type name ‘comp’
   46 | bigint *int_to_bi(BI_CTX *ctx, comp i);
      |                                ^~~~
make: *** [../../py/mkrules.mk:63: build-standard/extmod/modussl_axtls.o] Error 1
```

How exactly can we build micropython now using WSL?

markmelvin, 2020-12-18:
I also have my drives formatted as exFAT, if that matters.

markmelvin, 2020-12-18:
Looks like I may be better off converting my drive to NTFS...

Gary-Arcimoto, 2020-12-18:
I'm have the same issue. What can be done? Just NOT use the Windows mount and stick to all native Linux?

stinos, 2020-12-19:
> I also have my drives formatted as exFAT, if that matters.

exFAT doesn't support softlinks nor hardlinks so that is very likely the problem indeed. May I ask why you use exFat? 

> I'm have the same issue. What can be done?

Do you mean you also use exFAT? If not: see instructions wrt git and symlinks above.

markmelvin, 2020-12-19:
> exFAT doesn't support softlinks nor hardlinks so that is very likely the problem indeed. May I ask why you use exFat?

Seemed like a good idea at the time. I didn't realize it had these limitations. I've backed up and am reformatting as NTFS now and will try again when that is done.

markmelvin, 2020-12-19:
Switching to NTFS solved my problems. Thanks.

closed 2021-04-14 by dpgeorge