micropython/micropython #6522: ValueError: Pin(X1) doesn't exist

zcq111, 2020-10-07:
why can't use the Pin X1? 
the firmware is STM32F429DISC-20200930-unstable-v1.13-77-g0fd0eb00a.dfu, 
thanks.

>>> from pyb import Pin
>>> p_out = Pin('X1', Pin.OUT_PP)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin(X1) doesn't exist

zcq111, 2020-10-07:
>>> import lcd160cr
>>> 
>>> lcd = lcd160cr.LCD160CR('X')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lcd160cr.py", line 48, in __init__
ValueError: Pin(X4) doesn't exist
>>>

dpgeorge, 2020-10-07:
Those pins are not defined on that board.  To find out the pins do `dir(Pin.board)` or `dir(Pin.cpu)`.

zcq111, 2020-10-07:
I can't use the module lcd160cr, can't new the object lcd160cr.LCD160CR('X')
how to define those pins on the board?

dpgeorge, 2020-10-07:
> I can't use the module lcd160cr

You can specify explicit pins to the constructor, see http://docs.micropython.org/en/latest/library/lcd160cr.html?highlight=lcd160cr#lcd160cr.LCD160CR

zcq111, 2020-10-07:
thank you dpgeorge very much!

I want to know how to define those pins(X or Y) on the board, so I can dev my application by guider in example like :

lcd=lcd160cr.LCD160CR('X')

I can change / update and rebuild the project to create a new hex.

dpgeorge, 2021-04-14:
You need to define them in the boards `pins.csv` file.  See `ports/stm32/boards/PYBV11/pins.csv` for examples.

closed 2021-04-14 by dpgeorge