micropython/micropython #6523: Taking slice of memoryview in Viper function crashes PyBoard

twisteroidambassador, 2020-10-07:
On a PyBoard v1.1 with firmware 1.13, put the following code into a file:

    import micropython

    @micropython.viper
    def take_memoryview_slice():
        ba = bytearray(10)
        mv = memoryview(ba)
        slice = mv[0:1]

Running `take_memoryview_slice()` resets the PyBoard.

dpgeorge, 2020-10-07:
Thanks for the report, I can confirm the issue (on the unix port).

The problem is that the native emitter in viper mode cannot handle slices with literal integer arguments.  To work around it you can use:
```python
import micropython
from micropython import const

@micropython.viper
def take_memoryview_slice():
    ba = bytearray(10)
    mv = memoryview(ba)
    slice = mv[const(0):const(1)]
```

peterhinch, 2025-06-28:
This works (as per Damien's suggestion):
```py
from micropython import const

@micropython.viper
def memoryviewbug():

    ba = bytearray(16)
    mv = memoryview(ba)

    slc = mv[const(0): const(15)]
    print('OK')

memoryviewbug()
```
but this hangs:
```py
from micropython import const
a = const(0)
b = const(15)

@micropython.viper
def memoryviewbug():
  
    ba = bytearray(16)
    mv = memoryview(ba)
    slc = mv[a:b]
    print('OK')

memoryviewbug()
```
Tested with
```
MicroPython v1.25.0-72.g1b123579a.dirty on 2025-05-07; Raspberry Pi Pico with RP2040
```

agatti, 2025-07-02:
@RotatingLlama For what it's worth, this makes your test code work on RV32 and Arm under QEMU (not on x64, go figure):

```diff
diff --git i/py/emitnative.c w/py/emitnative.c
index f3ab483e8..e6872a8c3 100644
--- i/py/emitnative.c
+++ w/py/emitnative.c
@@ -2667,6 +2667,25 @@ static void emit_native_build_slice(emit_t *emit, mp_uint_t n_args) {
     if (n_args == 2) {
         vtype_kind_t vtype_start, vtype_stop;
         emit_pre_pop_reg_reg(emit, &vtype_stop, REG_ARG_2, &vtype_start, REG_ARG_1); // arg1 = start, arg2 = stop
+        if (vtype_start == VTYPE_INT || vtype_start == VTYPE_UINT) {
+            ASM_MOV_REG_REG(emit->as, REG_TEMP0, REG_ARG_2);
+            ASM_MOV_REG_IMM(emit->as, REG_ARG_2, vtype_start);
+            ASM_CALL_IND(emit->as, MP_F_CONVERT_NATIVE_TO_OBJ);
+            // REG_ARG_1 is now the wrapped start index
+            ASM_MOV_REG_REG(emit->as, REG_ARG_1, REG_RET);
+            ASM_MOV_REG_REG(emit->as, REG_ARG_2, REG_TEMP0);
+            vtype_start = VTYPE_PYOBJ;
+        }
+        if (vtype_stop == VTYPE_INT || vtype_stop == VTYPE_UINT) {
+            ASM_MOV_REG_REG(emit->as, REG_TEMP0, REG_ARG_1);
+            ASM_MOV_REG_REG(emit->as, REG_ARG_1, REG_ARG_2);
+            ASM_MOV_REG_IMM(emit->as, REG_ARG_2, vtype_stop);
+            ASM_CALL_IND(emit->as, MP_F_CONVERT_NATIVE_TO_OBJ);
+            // REG_ARG_2 is now the wrapped end index
+            ASM_MOV_REG_REG(emit->as, REG_ARG_2, REG_RET);
+            ASM_MOV_REG_REG(emit->as, REG_ARG_1, REG_TEMP0);
+            vtype_stop = VTYPE_PYOBJ;
+        }
         assert(vtype_start == VTYPE_PYOBJ);
         assert(vtype_stop == VTYPE_PYOBJ);
         emit_native_mov_reg_const(emit, REG_ARG_3, MP_F_CONST_NONE_OBJ); // arg3 = step
```

If that works for you on device then I'll clean this up (there's probably a better way to juggle the registers instead of what I do), extend it for the 3 args case as well, and then submit as a PR when I get it to work on x86/x64 as well.