micropython/micropython #6524: Space in wifi name

rajmasud, 2020-10-07:
If we have a space in the wifi name, the wlan.connect() wouldn't work. 
For an example, 
If we use wlan.connect('mywifi','mypassword') it will work but if the wifi name has a space as below:
wlan.connect('my wifi','mypassword'), it will fail. 

Is there any workaround?

dpgeorge, 2020-10-08:
What port/board are you using that's running MicroPython?

rajmasud, 2020-10-08:
I am using ESP32

dpgeorge, 2020-10-08:
MicroPython passes through the SSID to the underlying ESP-IDF, so if there's an issue connecting then it's most likely an IDF problem.  Searching for similar issues yields things like https://esp32.com/viewtopic.php?t=10185 .  It could be your router that is not compatible with ESP32.

kevinkk525, 2020-10-08:
That seems more likely. I have a space in my SSID and not problems with the ESP32.

rajmasud, 2020-10-08:
I was testing it with my mobile hotspot so that I can change the wifi name easily. Anytime I had a space, it failed. If I remove it, it works. I only have tested that far. Since Kevin has a space and it works, I'll try again and check my code to see if there is anything else I am missing. Thanks Damien and Kavin for your responses. I'll keep you posted with anything I find. Thanks.

jonnor, 2024-09-13:
Problem was not reproducible and no update in 3+ years. If problem persists, please re-test on latest MicroPython, provide details on how to reproduce and re-open the issue.

closed 2024-09-13 by jonnor