micropython/micropython #6538: machine.Pin doesn't have a pin() method on the ESP32 port

mattytrentini, 2020-10-10:
As [reported in the Slack channel](https://micropython.slack.com/archives/C485UCQHF/p1602356910072600), the ESP32 port lacks a method to retrieve the pin number. 

(It's possible to print the pin object but this is fragile and requires that the number be parsed from the string.)

The nRF and stm32 ports both have a `pin()` method on a `Pin`; the same should exist for the ESP32.

jonnor, 2024-09-13:
Is this still relevant @mattytrentini ?

mattytrentini, 2024-09-14:
It's likely to still be _accurate_, perhaps not _relevant_. @dpgeorge suggested that the `pin` shouldn't be necessary, there are easy workarounds. 

We should probably remove `pin` from nrf and stm32, assuming they're still present...