micropython/micropython #6539: Error trying to send a message to Telegram

mradovan, 2020-10-11:
There is tutorial how to send a message to telegram from python: https://creatronix.de/telegram-bot-with-micropython/

And to send a message to Telegram from MicroPython. In case of telegram, where I use requests to send a message to the Telegram:
import urequests as u
**req = u.get("https://api.telegram.org/bot<your_token>/sendMessage?text=hallo&chat_id=<your_chat_id>")**
, i get this:

**ssl_handshake_status: -40
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests.py", line 149, in get
  File "urequests.py", line 93, in request
OSError: [Errno 5] EIO**

If I use **"https://api.telegram.org/bot<your_token>/sendMessage?text=hallo&chat_id=<your_chat_id>"** directly in browser, the message is camming to my phone.

But, If in MicroPython I use for example this:
**req = u.get("https://telegram-webhook-sample.herokuapp.com/")**

It works correctly. 

My question is: what is wrong in my code?!

stinos, 2020-10-11:
This might be too broad to be on the MicroPython issue tracker, instead you should probably ask this to the people who created the blog or otherwise forum.micropython.org

schuhumi, 2021-04-13:
### TLS issue fix
I had the same problem, and it took me quite some time to figure out what's going on. For anyone else coming here from Google, this is what solved it for me: I use the unix port of micropython on an old wifi router, and the unix port uses axTLS by default, have a look here: https://github.com/micropython/micropython/blob/master/ports/unix/mpconfigport.mk
```
# ussl module requires one of the TLS libraries below
MICROPY_PY_USSL = 1
# axTLS has minimal size but implements only a subset of modern TLS
# functionality, so may have problems with some servers.
MICROPY_SSL_AXTLS = 1
# mbedTLS is more up to date and complete implementation, but also
# more bloated.
MICROPY_SSL_MBEDTLS = 0
```
It seems like telegram uses some TLS functionality not supported by axTLS, and switching to mbedTLS (just set AXTLS=0 and MBEDTLS=1) made it work fine (this configuration might differ for other ports, have a look at your specific case). So this whole thing at least for me is not actually a bug, its just unfortunate defaults for this usecase, and a rather useless error message.

### redirects fix in urequests
What you'll find next, if you use urequests, is that telegram uses redirects, and urequests doesn't support those ( see `raise NotImplementedError("Redirects not yet supported")` in https://github.com/micropython/micropython-lib/blob/master/urequests/urequests.py ). You can substitute it with https://github.com/pfalcon/pycopy-lib/blob/master/urequests/urequests/__init__.py for example, just make sure to adjust the SSL wrapping of the socket correctly (outdated? idk), not
```
ctx = ussl.SSLContext()
s = ctx.wrap_socket(s, server_hostname=host)
```
but rather
```
s = ussl.wrap_socket(s, server_hostname=host)
```

### Building with OpenWrt
In case you're using OpenWrt too and wondering how to get the correct SSL implementation into micropython when building: You'll find this file https://github.com/openwrt/packages/blob/master/lang/python/micropython/Makefile here `feeds/packages/lang/python/micropython/Makefile`. There change `DEPENDS:=+libffi` to `DEPENDS:=+libffi +libmbedtls`. Additionally create a file `feeds/packages/lang/python/micropython/patches/000-mbedtls.patch` that looks like this:
```
--- a/ports/unix/mpconfigport.mk
+++ b/ports/unix/mpconfigport.mk
@@ -28,11 +28,11 @@
 # axTLS has minimal size and fully integrated with MicroPython, but
 # implements only a subset of modern TLS functionality, so may have
 # problems with some servers.
-MICROPY_SSL_AXTLS = 1
+MICROPY_SSL_AXTLS = 0
 # mbedTLS is more up to date and complete implementation, but also
 # more bloated. Configuring and building of mbedTLS should be done
 # outside of MicroPython, it can just link with mbedTLS library.
-MICROPY_SSL_MBEDTLS = 0
+MICROPY_SSL_MBEDTLS = 1
 
 # jni module requires JVM/JNI
 MICROPY_PY_JNI = 0
```
to cater for the setting mentioned above. This exact patch will not fit anymore as soon as the mpconfigport.mk file changes, so better create a mpconfigport.mk.new (you find the file at `build_dir/<your target>/micropython-1.9.4/ports/<your port>/mpconfigport.mk` ), do the change, and create a patch with `diff -u mpconfigport.mk mpconfigport.mk.new` yourself (also make sure the --- a/... and +++ b/... paths look like in the example so that the buildsystem finds the correct file to patch). After that in your `make menuconfig` enable (so that there is a * in front, no M) Libraries->SSL->libmbedtls, and you should be good to go :)

jonnor, 2024-09-13:
Thanks @schuhumi for documenting a fix. Indeed it is critical with many web servers to use a modern TLS implementation.

closed 2024-09-13 by jonnor