micropython/micropython #6541: persistent ppp needed

emard, 2020-10-11:
Currently PPP tries to connect during some time, around 10 seconds
and the gives up. After the timeout LCP serial traffic is no longer active
at serial line.

In application I would need PPP to continuously be active
because if I restart PPP, it will spoil WiFi routing.

Additionaly - is this bug? This won't manually restart PPP after timeout:

ppp.active(False)
ppp.active(True)

Does nothing, while I think  it should reactivate PPP traffic to 
attempt connection on serial line

pumelo, 2020-10-17:
> then PPP will spoil WiFi routing

I'm using esp32 with ppp and an lte modem.
Whenever ppp fails I do call ppp.active(False) prior to creating a new instance of ppp. This seems to work fine. Do you see issues with this procedure?

> Additionaly - is this bug? This won't manually restart PPP after timeout:
> 
> ppp.active(False)
> ppp.active(True)

Reading the code in `ports/esp32/network_ppp.c` I don't see the reason why reusing the same instance should not work ... Do you see a reason? Are you calling `ppp.connect` after `ppp.active(True)` because this is required for sure.

However, I'm also throwing away my ppp instance once communication fails, but I can't recall the reason for doing so ...

emard, 2020-10-17:
Yes connect() follows of course.
Throwing ppp instance is not option here because
ppp is used in "reverse" functions, not as a way to go to internet
over GSM modem, but to provide internet for a small linux at ppp side,
while connected ESP32 to wifi

For this reason i can't "del ppp" instance and repoen it again because
WiFi will routing settings will be lost.

so ppp must be started first and be perisstent.
wifi must be started after ppp

pumelo, 2021-05-19:
BTW: Looks like this is an upstream issue https://github.com/espressif/esp-idf/issues/3506#issuecomment-842713341

chtinnes, 2021-09-15:
Some GSM modems support switching from data mode back to command mode and vice versa.
I would like to use this functionality to periodically exit data mode to get some GPS location from the modem.
Unfortunately, it looks like the command for switching to command mode is not working (maybe because there is some PPP communication ongoing). Is it possible to interrupt PPP and resuming it without closing existing sockets? If so, how?

jonnor, 2025-10-03:
Is this still relevant in recent MicroPython versions?