micropython/micropython #6546: can't install micropython

nesergen, 2020-10-14:
(LinuxMint18)

downloaded the latest master version micropython. Following the instructions, I go to the directory:
`$ cd ports/unix`

I execute the command:
`$ make submodules`

And I get the error:

```
 $ make submodulesUse make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/axtls lib/berkeley-db-1.xx lib/libffi
fatal: Not a git repository (or any of the parent directories): .git
../../py/mkrules.mk:186: ошибка выполнения рецепта для цели «submodules»
make: *** [submodules] Ошибка 128
```
what could be the problem?

dpgeorge, 2020-10-14:
> downloaded the latest master version micropython

Where did you download it from?  If you got the release version from micropython.org/download then there's no need to do `make submodules`, they are already included.

nesergen, 2020-10-14:
> Where did you download it from? If you got the release version from micropython.org/download then there's no need to do `make submodules`, they are already included.

I took the library from here:
https://github.com/micropython/micropython

nesergen, 2020-10-14:
> Where did you download it from? If you got the release version from micropython.org/download then there's no need to do `make submodules`, they are already included.

downloaded the library from "micropython.org/download". installed without problems.

closed 2020-10-14 by nesergen