micropython/micropython #6551: build error on linux-xtensa

cpackham, 2020-10-15:
Hi,

When mycropython building using buildroot we see the following error

```
/home/giuliobenetti/autobuild/run/instance-3/output-1/build/micropython-1.13/ports/unix/../../py/nlrxtensa.c:41:(.text.nlr_push+0x17): dangerous relocation: j: cannot encode: nlr_push_tail
collect2: error: ld returned 1 exit status
../../py/mkrules.mk:168: recipe for target 'micropython' failed
make[1]: *** [micropython] Error 1
```
[full output](http://autobuild.buildroot.net/results/ac7004261d9fa9aa8bc836a0303da2e43720a955/build-end.log)

I'm not sure if this is a problem in micropython or in the way the buildroot xtensa toolchain has been created. Can anyone shed any light on this issue.

dpgeorge, 2020-10-17:
It's probably because the distance to `nlr_push_tail` is greater that what can be encoded in a `j` instruction.  The fix would be to replace `j nlr_push_tail` with an indirect jump that can encode the entire address range, but I'm not exactly sure how to do that.  Alternatively you could try compiling with `MICROPY_NLR_SETJMP` enabled instead of `MICROPY_NLR_XTENSA`.

jcmvbkbc, 2021-03-29:
> The fix would be to replace j nlr_push_tail with an indirect jump 

There's another issue in this source file: it assumes call0 ABI, whereas linux toolchain is most likely to be configured for the windowed ABI. Jumping directly to a function symbol and returning from regular function with `ret.n` will not work in windowed ABI.

jonnor, 2024-09-13:
Is this issue still relevant on recent MicroPython versions?

closed 2025-03-03 by jonnor