micropython/micropython #6553: ESP32 execute esp.read_flash() cause exception

Walkline80, 2020-10-16:
* firmware version: MicroPython v1.13-106-g893f75546-dirty on 2020-10-16
* customize partition table:

    | name | type | sub-type | offset | size |
    | :-: | :-: | :-: | :-: | :-: |
    | nvs | data | nvs | 0x9000 | 0x6000 |
    | phy_init | data | phy | 0xf000 | 0x1000 |
    | factory | app | factory | 0x10000 | 0x180000 |
    | vfs | data | fat | 0x200000 | 0x1ff000 |
    | user_data | data | fat | 0x3ff000 | 0x1000 |

* executed code:
    ```
    >>> import esp32
    >>> part=esp32.Partition
    >>> part.find(type=part.TYPE_DATA)
    [<Partition type=1, subtype=2, address=36864, size=24576, label=nvs, encrypted=0>, <Partition type=1, subtype=1, address=61440, size=4096, label=phy_init, encrypted=0>, <Partition type=1, subtype=129, address=2097152, size=2093056, label=vfs, encrypted=0>, <Partition type=1, subtype=129, address=4190208, size=4096, label=user_data, encrypted=0>]
    >>> user=part("user_data")
    >>> user.info()
    (1, 129, 4190208, 4096, 'user_data', False)
    >>>
    >>>
    >> import esp
    >>> esp.flash_size()
    4194304
    >>> esp.flash_write(0x3ff000, b'walkline')
    >>> esp.flash_read(0x3ff000, 0x10)
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    TypeError: object with buffer protocol required
    >>>
    ```

According to the [doc](http://docs.micropython.org/en/latest/library/esp.html#esp.flash_read), the 2nd param is a length of buffer that I want to read, but why it raise a TypeError to me?

BTW, are there anyone knows how to use `esp32.Partition` to read and write blocks?

Thanks~

Walkline80, 2020-10-16:
I found the difference between ESP32 and ESP8266, ESP8266 can directly pass in the length param, but ESP32 have to pass in a `bytearray` variable

```
>>> import esp
>>> buff=bytearray(8)
>>> esp.flash_read(0x3ff000, buff)
>>> buff
bytearray(b'walkline')
```

dpgeorge, 2021-04-14:
This needs an update to the docs to state that esp32 doesn't support passing a length, just buffer.