micropython/micropython #6556: ESP32: Arming timer inside `schedule`

pulkin, 2020-10-19:
I cannot properly invoke timer. The idea is to invoke vibro attached to `4` of ESP32 such that it will be turned off after a short period. What happens by fact is that the 'off' callback is only invoked after some unexpected time. It looks like all `schedule`d tasks are done with interrupts disabled (?). If it is the case than it is not the desired behaviour for whoever writes event-driven code with `schedule`.
```python
vibro = Pin(4, OUT)

class Vibro:
    @staticmethod
    def off(source):
        vibro.off()
    @staticmethod
    def pulse(time):
        vibro.on()
        Timer(3).init(period=time, mode=Timer.ONE_SHOT, callback=Vibro.off)

def please_do_a_short_pulse(*args):
    Vibro.pulse(100)
    sleep_ms(1000)

schedule(please_do_a_short_pulse, None)
sleep_ms(1000)
```
In the end, how should I properly implement `Vibro`?

dpgeorge, 2020-10-20:
> It looks like all `schedule`d tasks are done with interrupts disabled

Yes that is correct, in a sense.  A function run via a `micropython.schedule()` is executed in a context so no other scheduled callbacks are made.  Scheduled callbacks are only made when "thread level"/"top level" code is executing.  This is to make scheduling fair (FIFO), non-recursive, and give guarantees when executing a scheduled function (like shared access to a common resource).

> In the end, how should I properly implement `Vibro`?

Don't call sleep in a scheduled function.  Scheduled functions/callbacks should run as fast as possible and not block.

peterhinch, 2020-10-20:
@pulkin An option is to use **uasyncio** instead of timers. [Official docs](http://docs.micropython.org/en/latest/library/uasyncio.html) and [unofficial tutorial](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md).

pulkin, 2020-10-20:
I have a solid piece of code reading and processing some file in place of `sleep_ms`. I want to implement the following logic: `press a button (interrupt) -> start vibro for tactile feedback -> run some workload`. Vibro has to shut down after 0.1s via timer interrupt.

It looks like I need to run some sort of main loop processing a queue of tasks initiated by interrupts. Is there any reference implementation for it?

dpgeorge, 2020-10-22:
> It looks like I need to run some sort of main loop processing a queue of tasks initiated by interrupts. Is there any reference implementation for it?

You can use `uasyncio`, combined with a timer to turn off the vibro after the desired time.  Or set a flag in the button IRQ handler for the main loop to see that it should run the workload.

Otherwise, this is not really a bug report so I'll close this issue and discussion can continue at https://forum.micropython.org if needed.

closed 2020-10-22 by dpgeorge