micropython/micropython #6557: Feature request: BLE HID

street-grease-coder, 2020-10-20:
Analogous to the Arduino code below, it would make a compelling library/feature in the micropython arsenal. Because it is not available in any form (and goes so well together with IMUs to create an easy BLE mouse / air mouse (and ESP32s are extremely abundant and cheap)) I have to switch to Arduino code again from micropython (also, because I'm a noob). 

Anyway, if there is space for this - I think it'd make micropython more awesome, and it's the only thing standing in the way of building super easy game controllers.

Working library for arduino here, all credits to the authors
https://github.com/T-vK/ESP32-BLE-Mouse

Walkline80, 2020-10-20:
In fact, you can combine the necessary services of the HID device by yourself to implement HID master control devices such as mouse and keyboard. For example, I implemented a BLE keyboard with only one button (using the boot button on the ESP32 development board), but it is not yet supports device pairing and binding, so it cannot be used on Windows and iOS.

street-grease-coder, 2020-10-20:
I'm a bit puzzled how to go about this. I realize, after pairing it's essentially just sending the right code for 'left' to move the mouse left. But usually HID assumes native USB support - in the case of ESP32 this is notoriously not the case though and this complicates things a lot more (to my, limited, understanding) and making a library super useful

Walkline80, 2020-10-20:
> I'm a bit puzzled how to go about this. I realize, after pairing it's essentially just sending the right code for 'left' to move the mouse left. But usually HID assumes native USB support - in the [...]

I don't quite understand what you means, so I can only say that there is no easy way to achieve your requests (similar to Arduino)

dpgeorge, 2020-10-20:
Please see #6559 for some work-in-progress code to implement BLE HID mouse and keyboard devices.

street-grease-coder, 2020-10-20:
I think this might be exactly what I was looking for...... amazing, this was posted today! will test

street-grease-coder, 2020-10-20:
@dpgeorge under windows 10 you can connect to it as mouse, but send_motion() has no effect (unless I messed it up). If you give me some clues I could read into how to fix/adapt

Walkline80, 2020-10-20:
> @dpgeorge under windows 10 you can connect to it as mouse, but send_motion() has no effect (unless I messed it up). If you give me some clues I could read into how to fix/adapt
> 
> I'm basically running this:
> 
> ```
> import ble_hid_mouse as mouse
> while True:
>   mouse.send_motion(20,10)
> ```
> 
> ps: [this might help](https://gitee.com/walkline/esp32-ble/blob/master/ble/ble_hid.py)

Please note that currently BLE HID only supports the use of android, and there is no automatic reconnect function, yet

And, the link you mentioned used additional modules, which cannot be used directly for you.

street-grease-coder, 2020-10-20:
I see... ok, I think I would have to know much more about BLE and micropython to be useful here

street-grease-coder, 2020-10-26:
@Walkline80 you could have mentioned that you have an entire public repo with your some example code online on gitee ^^

street-grease-coder, 2020-10-27:
I played around with the code a bit (added print commands, stuff below), however I can definitely confirm that it does not work under windows with esp32 port of micropython idf 1.4. 

How do I go about debugging this best? until I understood that `struct.pack("4B",...)` means 'pack 4 bytecodes' into the struct, it took a while... I was looking for more resources and example code for this, but it's seemingly really hard to find. Are there any resources to read up on to understand this communication better, and how to code this using micropython?

Is there any way to reconnect to the device automatically or has this not been implemented yet in micropython? 

[image]
[image]

```
# Implements a BLE HID mouse

from micropython import const
import struct
import bluetooth
import time

isconnected=False
isready=False

def ble_irq(event, data):
    global conn_handle
    global isconnected, isready
    if event == 1:
        isconnected=True
        print("connect: ",event,data)
        conn_handle = data[0]
    elif event ==2 and isconnected:
        isconnected=False
        print("Disconnect", event, data)
        #gnah... cannot reconnect...
        conn_handle = data[0]
    else:
        print("other events:", event, data)
        conn_handle = data[0]

ble = bluetooth.BLE()
ble.active(1)
ble.irq(ble_irq)

UUID = bluetooth.UUID

F_READ = bluetooth.FLAG_READ
F_WRITE = bluetooth.FLAG_WRITE
F_READ_WRITE = bluetooth.FLAG_READ | bluetooth.FLAG_WRITE
F_READ_NOTIFY = bluetooth.FLAG_READ | bluetooth.FLAG_NOTIFY

ATT_F_READ = 0x01
ATT_F_WRITE = 0x02

hid_service = (
    UUID(0x1812),  # Human Interface Device
    (
        (UUID(0x2A4A), F_READ),  # HID information
        (UUID(0x2A4B), F_READ),  # HID report map
        (UUID(0x2A4C), F_WRITE),  # HID control point
        (UUID(0x2A4D), F_READ_NOTIFY, ((UUID(0x2908), ATT_F_READ),)),  # HID report / reference
        (UUID(0x2A4E), F_READ_WRITE),  # HID protocol mode
    ),
)

# fmt: off
HID_REPORT_MAP = bytes([
    0x05, 0x01,     # Usage Page (Generic Desktop)
    0x09, 0x02,     # Usage (Mouse)
    0xA1, 0x01,     # Collection (Application)
    0x09, 0x01,     #     Usage (Pointer)
    0xA1, 0x00,     #     Collection (Physical)
    0x85, 0x01,     #         Report ID (1)
    0x95, 0x03,     #         Report Count (3)
    0x75, 0x01,     #         Report Size (1)
    0x05, 0x09,     #         Usage Page (Buttons)
    0x19, 0x01,     #         Usage Minimum (1)
    0x29, 0x03,     #         Usage Maximum (3)
    0x15, 0x00,     #         Logical Minimum (0)
    0x25, 0x01,     #         Logical Maximum (1)
    0x81, 0x02,     #         Input(Data, Variable, Absolute); 3 button bits
    0x95, 0x01,     #         Report Count(1)
    0x75, 0x05,     #         Report Size(5)
    0x81, 0x01,     #         Input(Constant);                 5 bit padding
    0x75, 0x08,     #         Report Size (8)
    0x95, 0x02,     #         Report Count (3)
    0x05, 0x01,     #         Usage Page (Generic Desktop)
    0x09, 0x30,     #         Usage (X)
    0x09, 0x31,     #         Usage (Y)
    0x09, 0x38,     #         Usage (Wheel)
    0x15, 0x81,     #         Logical Minimum (-127)
    0x25, 0x7F,     #         Logical Maximum (127)
    0x81, 0x06,     #         Input(Data, Variable, Relative); 3 position bytes (X,Y,Wheel)
    0xC0,           #     End Collection
    0xC0,           # End Collection
])
# fmt: on

# register services
ble.config(gap_name="MP-mouse")
handles = ble.gatts_register_services((hid_service,))
print(handles)
h_info, h_hid, _, h_rep, h_d1, h_proto = handles[0]

# set initial data
ble.gatts_write(h_info, b"\x01\x01\x00\x02")     # HID info: ver=1.1, country=0, flags=normal
ble.gatts_write(h_hid, HID_REPORT_MAP)           # HID report map
ble.gatts_write(h_d1, struct.pack("<BB", 1, 1))  # report: id=1, type=input
ble.gatts_write(h_proto, b"\x01")                # protocol mode: report

# advertise
adv = (
    b"\x02\x01\x06"
    b"\x03\x03\x12\x18"  # complete list of 16-bit service UUIDs: 0x1812
    b"\x03\x19\xc2\x03"  # appearance: mouse
    b"\x09\x09MP-mouse"  # complete local name
)
conn_handle = None
ble.gap_advertise(100_000, adv)

# once connected use the following to send reports

def send_mouse(button_mask, x, y, wheel):
    ble.gatts_notify(conn_handle, h_rep, struct.pack("4B", button_mask, x, y, wheel))

def send_click(button):
    send_mouse(1 << button, 0, 0, 0)
    send_mouse(0, 0, 0, 0)

def send_motion(x, y):
    send_mouse(0, x, y, 0)

for x in range(0,20):
    print("loop: " + str(isconnected))
    if isconnected:
        print("isconnected, moving mouse...")
        send_motion(1,1)
    time.sleep(1)

```

street-grease-coder, 2020-10-31:
Suggestion: Can this be merged with BLE additional features sticky? I think it would be a major step forward to control your computer with micropython effectively and wirelessly. I'll close this then

closed 2021-01-27 by street-grease-coder

Josverl, 2025-12-24:
> I'd like to process input of a bluetooth keyboard.

@LTek-online, If you want to request a feature then please open a new Discussion or issue rather than commenting on a closed one. Especially as you seem to be asking about different functionality

You can add a link to this, and any other relevant PRs to keep the context.