micropython/micropython #6564: GSM PPP is not Attaching to Micropython network.PPP

vijenderpanda, 2020-10-22:
Kindly check ,

class InitSimData:
    def __init__(self):
        self.gsm = machine.UART(1,tx=2, rx=4, timeout=1000,  baudrate=9600)
        time.sleep(interval)
        self.check_sim()

        time.sleep(interval)
        self.register_sim()

        # time.sleep(interval)
        # self.getOPR()

        time.sleep(interval)
        self.attachGPRS()

        time.sleep(interval)
        self.rquestIPwithAPN()

        time.sleep(interval)
        self.getCGDCONT()

        # time.sleep(interval)
        # self.qosProfile()

        # time.sleep(interval)
        # self.qosProfilenext()

        time.sleep(interval)
        self.checkIP()

        time.sleep(interval)
        self.PDPcontext()

        time.sleep(interval)
        self.attachGSMtoPPP()

        

        
        
        

        
        

        

    def check_sim(self):
        while True:
            self.gsm.write("AT\r\n")
            status = self.read_status()
            if status == "OK":
                print("Sim Initialized with status with status",status)
                return True
            else:
                pass
    def register_sim(self):
        while True:
            self.gsm.write("AT+CREG?\r\n")
            status = self.read_status()
            if status == "+CREG: 0,1":
                print("Sim Registered with status with status",status)
                return True
            else:
                pass
    def getOPR(self):
        while True:
            self.gsm.write("AT+COPS?\r\n")
[... 153 lines cut ...]
b'\r\nOK\r\n'                                                                                                                                         
>>> gsm.write('AT+CGDCONT=1,"IP","portalnmms"\r\n')                                                                                                   
32                                                                                                                                                    
>>> print(gsm.read())                                                                                                                                 
b'\r\nOK\r\n'                                                                                                                                         
>>> gsm.write('AT+CGDATA="PPP",1\r\n')                                                                                                                
19                                                                                                                                                    
>>> print(gsm.read())                                                                                                                                 
b'\r\nERROR\r\n'     

>>> gsm.write('ATD*99***1#\r\n')                                                                                                                      
13                                                                                                                                                    
>>> print(gsm.read())                                                                                                                                 
b'\r\n\r\nCONNECT\r\n\r\n' 

>>> GPRS.isconnected()                                                                                                                                
False                                                                                                                                                 
>>> GPRS.isconnected()                                                                                                                                
False                                                                                                                                                 
>>> GPRS.isconnected()                                                                                                                                
False                                                                                                                                                 
>>> GPRS.isconnected()                                                                                                                                
False                                                                                                                                                 
>>> GPRS.isconnected()                                                                                                                                
False

vijenderpanda, 2020-10-22:
> Kindly check ,
> i followed to your last answer that we need to pass the authmode while connecting,
> i have don that too , but still no luck
> 
> i am trying on ESP32 and SIM900A
> 
> class InitSimData():
> def **init**(self):
> self.gsm = machine.UART(1,tx=2, rx=4, timeout=1000, baudrate=9600)
> time.sleep(interval)
> self.check_sim()
> 
> ```
>     time.sleep(interval)
>     self.register_sim()
> 
>     # time.sleep(interval)
>     # self.getOPR()
> 
>     time.sleep(interval)
>     self.attachGPRS()
> 
>     time.sleep(interval)
>     self.rquestIPwithAPN()
> 
>     time.sleep(interval)
>     self.getCGDCONT()
> 
>     # time.sleep(interval)
>     # self.qosProfile()
> 
>     # time.sleep(interval)
>     # self.qosProfilenext()
> 
> 
>     time.sleep(interval)
>     self.checkIP()
> 
>     time.sleep(interval)
>     self.PDPcontext()
> 
> 
>     time.sleep(interval)
>     self.attachGSMtoPPP()
> 
> 
>     
> 
>     
>     
>     
> 
>     
>     
> 
>     
> 
> 
> def check_sim(self):
>     while True:
>         self.gsm.write("AT\r\n")
>         status = self.read_status()
>         if status == "OK":
>             print("Sim Initialized with status with status",status)
>             return True
>         else:
>             pass
> def register_sim(self):
>     while True:
>         self.gsm.write("AT+CREG?\r\n")
[... 166 lines cut ...]
> > > > b'\r\nOK\r\n'
> > > > gsm.write('AT+CGDCONT=1,"IP","portalnmms"\r\n')
> > > > 32
> > > > print(gsm.read())
> > > > b'\r\nOK\r\n'
> > > > gsm.write('AT+CGDATA="PPP",1\r\n')
> > > > 19
> > > > print(gsm.read())
> > > > b'\r\nERROR\r\n'
> 
> > > > gsm.write('ATD_99_**1#\r\n')
> > > > 13
> > > > print(gsm.read())
> > > > b'\r\n\r\nCONNECT\r\n\r\n'
> 
> > > > GPRS.isconnected()
> > > > False
> > > > GPRS.isconnected()
> > > > False
> > > > GPRS.isconnected()
> > > > False
> > > > GPRS.isconnected()
> > > > False
> > > > GPRS.isconnected()
> > > > False

abuvanth, 2023-01-12:
here is the working code for jio sim

https://gist.github.com/abuvanth/001aa1cb28617faff9c3ab72409dd20a