micropython/micropython #6566: uasyncio raises "local variable referenced before assignment"

amirgon, 2020-10-23:
When trying to cancel a task that is waiting on `wait_for` with timeout, `wait_for` tries to access `ret` although it is not assigned.

Example:

```python
import uasyncio

async def task1():
    await uasyncio.sleep(3)

async def task2(t1):
    await uasyncio.sleep(1)
    t1.cancel()

async def task3(t1):
    try:
        await uasyncio.wait_for(t1, 2)
    except uasyncio.TimeoutError:
        pass

t1 = uasyncio.create_task(task1())
t2 = uasyncio.create_task(task2(t1))
t3 = uasyncio.create_task(task3(t1))
uasyncio.Loop.run_forever()
```

Result (unix port "dev" variant):
```
Task exception wasn't retrieved
future: <Task> coro= <generator object 'task3' at 7fae42272b00>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 12, in task3
  File "uasyncio/funcs.py", line 1, in wait_for
NameError: local variable referenced before assignment
```

dpgeorge, 2020-10-29:
See related #5797

amirgon, 2020-10-29:
Duplicate of #5797.
Closing.

closed 2020-10-29 by amirgon

dpgeorge, 2020-12-02:
Fixed by b505971069333a373d9f0c12138b64dab83fed72