micropython/micropython #6568: ESP32: littlefs - Guru MEditation Error - Reboot - lfs2_pair_cmp at lfs2.c

straga, 2020-10-24:
MicroPython v1.13-103-gb137d064e-dirty on 2020-10-23; ANY with ESP32

If use littlefs: reboot.
But the same code with fatfs works fine.

Code Run in thread asyncio and a lot read-write files from flash.

---
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401d8123  PS      : 0x00060730  A0      : 0x80101972  A1      : 0x3ffd7e10
0x401d8123: lfs2_pair_cmp at lfs2.c:?

A2      : 0xe613b061  A3      : 0x3f82d3b8  A4      : 0x00000000  A5      : 0x3f82c858
A6      : 0x00000001  A7      : 0x00001ba2  A8      : 0x00000001  A9      : 0x00000001
A10     : 0xec4fa8bb  A11     : 0x00000001  A12     : 0x13b05968  A13     : 0x00000224
A14     : 0x00000001  A15     : 0x00000522  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c
EXCVADDR: 0xe613b061  LBEG    : 0x40094624  LEND    : 0x40094652  LCOUNT  : 0x00000000
0x40094624: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x40094652: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 0000000000000000

Backtrace: 0x401d8120:0x3ffd7e10 0x4010196f:0x3ffd7e30 0x401020fe:0x3ffd7ee0 0x40102125:0x3ffd7f20 0x400f6e32:0x3ffd7f40 0x400ea505:0x3ffd7f60 0x400f6da5:0x3ffd7f90 0x400e4042:0x3ffd7fb0 0x400dfd7d:0x3ffd7fe0 0x400dfeb9:0x3ffd8000 0x400ed6f5:0x3ffd8020 0x400e4104:0x3ffd80c0 0x400dfd7d:0x3ffd80f0 0x400dfeb9:0x3ffd8110 0x400ede93:0x3ffd8130 0x400e4104:0x3ffd81d0 0x400dfd7d:0x3ffd8250 0x400e25e0:0x3ffd8270 0x400e25fd:0x3ffd82c0 0x400dfd7d:0x3ffd82e0 0x400e04d4:0x3ffd8300 0x400ede4d:0x3ffd8360 0x400e4104:0x3ffd8400 0x400dfd7d:0x3ffd8430 0x400e04d4:0x3ffd8450 0x400edee1:0x3ffd84b0 0x400e4447:0x3ffd8550 0x400dffe1:0x3ffd8570 0x400ee09d:0x3ffd85b0 0x400e4447:0x3ffd8650 0x400dffe1:0x3ffd8670 0x400ee09d:0x3ffd86b0 0x400e4447:0x3ffd8750 0x400e44b9:0x3ffd8770 0x400e4528:0x3ffd87a0 0x400e3fad:0x3ffd87c0 0x400dfd7d:0x3ffd87e0 0x400dfeb9:0x3ffd8800 0x400ede93:0x3ffd8820 0x400e4104:0x3ffd88c0 0x400dfd7d:0x3ffd88f0 0x400ede0d:0x3ffd8910 0x400e4104:0x3ffd89b0 0x400dfd7d:0x3ffd8a10 0x400ecec8:0x3ffd8a30 0x400fbf79:0x3ffd8ac0 0x4009690d:0x3ffd8ae0
0x401d8120: lfs2_pair_cmp at lfs2.c:?

0x4010196f: lfs2_dir_commit at lfs2.c:?

0x401020fe: lfs2_file_sync at ??:?

0x40102125: lfs2_file_close at ??:?

0x400f6e32: mp_vfs_lfs2_file_ioctl at vfs_lfs.c:?

0x400ea505: mp_stream_close at ??:?

0x400f6da5: mp_vfs_lfs2_file___exit__ at vfs_lfs.c:?

0x400e4042: fun_builtin_var_call at objfun.c:?

0x400dfd7d: mp_call_function_n_kw at ??:?

0x400dfeb9: mp_call_method_n_kw at ??:?

0x400ed6f5: mp_execute_bytecode at ??:?

0x400e4104: fun_bc_call at objfun.c:?

0x400dfd7d: mp_call_function_n_kw at ??:?

0x400dfeb9: mp_call_method_n_kw at ??:?

0x400ede93: mp_execute_bytecode at ??:?

0x400e4104: fun_bc_call at objfun.c:?

0x400dfd7d: mp_call_function_n_kw at ??:?

0x400e25e0: mp_call_method_self_n_kw at ??:?

0x400e25fd: bound_meth_call at objboundmeth.c:?

0x400dfd7d: mp_call_function_n_kw at ??:?

0x400e04d4: mp_call_method_n_kw_var at ??:?

0x400ede4d: mp_execute_bytecode at ??:?

0x400e4104: fun_bc_call at objfun.c:?

0x400dfd7d: mp_call_function_n_kw at ??:?

0x400e04d4: mp_call_method_n_kw_var at ??:?

0x400edee1: mp_execute_bytecode at ??:?

0x400e4447: mp_obj_gen_resume at ??:?

0x400dffe1: mp_resume at ??:?

0x400ee09d: mp_execute_bytecode at ??:?

0x400e4447: mp_obj_gen_resume at ??:?

0x400dffe1: mp_resume at ??:?

0x400ee09d: mp_execute_bytecode at ??:?

0x400e4447: mp_obj_gen_resume at ??:?

0x400e44b9: gen_resume_and_raise at objgenerator.c:?

0x400e4528: gen_instance_send at objgenerator.c:?

0x400e3fad: fun_builtin_2_call at objfun.c:?

0x400dfd7d: mp_call_function_n_kw at ??:?

0x400dfeb9: mp_call_method_n_kw at ??:?

0x400ede93: mp_execute_bytecode at ??:?

0x400e4104: fun_bc_call at objfun.c:?

0x400dfd7d: mp_call_function_n_kw at ??:?

0x400ede0d: mp_execute_bytecode at ??:?

0x400e4104: fun_bc_call at objfun.c:?

0x400dfd7d: mp_call_function_n_kw at ??:?

0x400ecec8: thread_entry at modthread.c:?

0x400fbf79: freertos_entry at mpthreadport.c:?

0x4009690d: vPortTaskWrapper at port.c:?

Rebooting...
---

straga, 2020-12-13:
with littlefs2 to v2.3.0. Same result. With Fat work ok.

straga, 2021-02-26:
26.02.2021 same, littlefs not works, vfat works.

```
DEBUG:FJSON:fullpath: ./u_config/_schema/mqtt_cfg
DEBUG:FJSON:Data dump !
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401cc933  PS      : 0x00060530  A0      : 0x800f2686  A1      : 0x3ffcecb0
0x401cc933: lfs2_pair_cmp at /opt/upy/micropython/lib/littlefs/lfs2.c:267

A2      : 0x15122508  A3      : 0x3f828578  A4      : 0x00000002  A5      : 0x00000001
A6      : 0x00000500  A7      : 0x3f8178a8  A8      : 0x00000001  A9      : 0x00000001
A10     : 0x00000001  A11     : 0xeaedddcc  A12     : 0x15122500  A13     : 0x000002cd
A14     : 0x000002cc  A15     : 0x000000a4  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x15122508  LBEG    : 0x40092a75  LEND    : 0x40092a85  LCOUNT  : 0xfffffffe
0x40092a75: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84

0x40092a85: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

ELF file SHA256: 40d6efd9c65c93c5

Backtrace: 0x401cc930:0x3ffcecb0 0x400f2683:0x3ffcecd0 0x400f3252:0x3ffced80 0x400f39d9:0x3ffcedd0 0x400ef659:0x3ffcedf0 0x400ddb0d:0x3ffcee10 0x400e4139:0x3ffcee30 0x400e428a:0x3ffcee50 0x400ee005:0x3ffcee70 0x400ee2c6:0x3ffceea0 0x400ddb5e:0x3ffceed0 0x400e4139:0x3ffcef00 0x400e68f2:0x3ffcef20 0x400ddbf8:0x3ffcefc0 0x400e4139:0x3ffceff0 0x400e428a:0x3ffcf010 0x400e69a1:0x3ffcf030 0x400ddbf8:0x3ffcf0d0 0x400e4139:0x3ffcf140 0x400e428a:0x3ffcf160 0x400e69a1:0x3ffcf180 0x400ddbf8:0x3ffcf220 0x400e4139:0x3ffcf250 0x40105df2:0x3ffcf270 0x40105e25:0x3ffcf2b0 0x400e4139:0x3ffcf2d0 0x400e4806:0x3ffcf2f0 0x400e6949:0x3ffcf350 0x400ddbf8:0x3ffcf3f0 0x400e4139:0x3ffcf420 0x400e4806:0x3ffcf440 0x400e69f5:0x3ffcf4a0 0x400ddf74:0x3ffcf540 0x400e43bb:0x3ffcf560 0x400e6b74:0x3ffcf5a0 0x400ddf74:0x3ffcf640 0x400e43bb:0x3ffcf660 0x400e6b74:0x3ffcf6a0 0x400ddf74:0x3ffcf740 0x400ddfc0:0x3ffcf760 0x400de030:0x3ffcf790 0x400ddad1:0x3ffcf7b0 0x400e4139:0x3ffcf7d0 0x400e428a:0x3ffcf7f0 0x400e69a1:0x3ffcf810 0x400ddbf8:0x3ffcf8b0 0x400e4139:0x3ffcf8e0 0x400e68f2:0x3ffcf900 0x400ddbf8:0x3ffcf9a0 0x400e4139:0x3ffcf9f0 0x401044ef:0x3ffcfa10 0x400d94bd:0x3ffcfaa0
0x401cc930: lfs2_pair_cmp at /opt/upy/micropython/lib/littlefs/lfs2.c:266

0x400f2683: lfs2_dir_commit at /opt/upy/micropython/lib/littlefs/lfs2.c:1829 (discriminator 1)

0x400f3252: lfs2_file_rawopencfg at /opt/upy/micropython/lib/littlefs/lfs2.c:2503

0x400f39d9: lfs2_file_opencfg at /opt/upy/micropython/lib/littlefs/lfs2.c:5076

0x400ef659: mp_vfs_lfs2_file_open at /opt/upy/micropython/extmod/vfs_lfsx_file.c:116
```

dpgeorge, 2021-03-11:
If you can provide Python code which fully reproduces this issue that would help to diagnose it.  Alternatively you might be able to dump the block device/partition that littlefs is on and get it to crash reliably on that.

straga, 2021-03-11:
> If you can provide Python code which fully reproduces this issue that would help to diagnose it. Alternatively you might be able to dump the block device/partition that littlefs is on and get it to crash reliably on that.

OK. I do full code how what I do. Couple days.

thyrrestrup, 2021-03-11:
lalala

straga, 2021-03-12:
1. board

Chip is ESP32D0WDQ5 (revision 1) PSRAM 4MB
flash 8MB

partitions:
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
ota_0,app,ota_0,0x10000,2100K,
ota_1,app,ota_1,0x220000,2100K,
vfs,data,fat,0x42d000,3100K,

ESP-IDF v4.1.1
git: https://github.com/straga/micropython_littlefs_test/tree/master/data_esp
Today build from source. Just put files from git on board. First boot will be froze and not hapen, after reset will be

[image]

dpgeorge, 2021-03-13:
Thank you.  I'm able to reproduce the crash in lfs2_pair_cmp.

dpgeorge, 2021-04-14:
I did find the root cause of this and thought I posted the outcome here, but it looks like the comment did not get posted.

The problem is that `ilistdir()` is being used without exhausting the iterator, and so the open directory descriptor never gets closed, even though the GC reclaims and reuses the memory.  The fix would be to make the `ilistdir()` iterator have a finaliser.

In the meantime, a workaround is to always exhaust the iterator, eg with `list(ilistdir())`.

Minimal code to reproduce the issue (on any port):
```python
import gc

try:
    import uos

    uos.VfsLfs2
except (ImportError, AttributeError):
    print("SKIP")
    raise SystemExit

class RAMBlockDevice:
    ERASE_BLOCK_SIZE = 1024

    def __init__(self, blocks):
        self.data = bytearray(blocks * self.ERASE_BLOCK_SIZE)

    def readblocks(self, block, buf, off):
        addr = block * self.ERASE_BLOCK_SIZE + off
        for i in range(len(buf)):
            buf[i] = self.data[addr + i]

    def writeblocks(self, block, buf, off):
        addr = block * self.ERASE_BLOCK_SIZE + off
        for i in range(len(buf)):
            self.data[addr + i] = buf[i]

    def ioctl(self, op, arg):
        if op == 4:  # block count
            return len(self.data) // self.ERASE_BLOCK_SIZE
        if op == 5:  # block size
            return self.ERASE_BLOCK_SIZE
        if op == 6:  # erase block
            return 0

def test(bdev, vfs_class):
    vfs_class.mkfs(bdev)
    vfs = vfs_class(bdev)

    for i in range(10):
        print(i)
        l = vfs.ilistdir("/")
        #list(l)  # uncomment and the bug is no longer there because the iterator is exhausted
        gc.collect()
        vfs.open("/test", "w").close()

bdev = RAMBlockDevice(30)
test(bdev, uos.VfsLfs2)
```

straga, 2021-04-20:
Thanks a lot. Now works.

dpgeorge, 2021-04-20:
Glad to hear the workaround works.

But this needs to be fixed on the C side, so reopening.

dpgeorge, 2022-09-13:
Fixed by 4e0964b59f44b25ebaa5239f9ea4273b804ebe64

closed 2022-09-13 by dpgeorge