micropython/micropython #6570: _IRQ_GATTC_NOTIFY  not compatible

arthurj58, 2020-10-25:
Hello there 
I am working on microPython BLE support
The _IRQ_GATTC_NOTIFY function of Bluetooth ble_temperature.py is not compatible
The physical device is connected to nrf51822 uart.
I installed LightBlue -> Virtual Device -> HeartRate -> 180D -> 2A37 microPython NOTIFY on iPhone not working
mycropython ubluetooth.BLE enable "Listen for notifications" function?

EPS-WROOM-32
esp32-idf4-20201024-unstable-v1.13-126-g05f95682e
thank

arthurj58, 2020-10-25:
[image]
Only "Notify" No "Read"
Notify event cannot work

jimmo, 2020-10-26:
Hi,
Please include some example code for the GATT client showing how you're connecting to the peripheral and receiving the notify event. Have you subscribed to notifications on the remote device?

arthurj58, 2020-10-27:
1 . Run ble_simple_peripheral.py 

2. Open IOS LightBlue App
	a.find EPS32 Peripheral
	b.connect to EPS32
	c.click Clone
	d.delete 6E400003-B5A3-F393-E0A9-E50E24DCCA9E properties “Read” property 
	e. Stop ble_simple_peripheral.py , 

3. Now create a virtual device, We use another phone to connect , Test Notify has a value

Video:  "https://youtu.be/klu6IezMzPs"

4. Open  ble_simple_central.py 
	Test from iPhone LightBlue App Notify data

Can't get any data

Code : 
`# This example finds and connects to a peripheral running the
# UART service (e.g. ble_simple_peripheral.py).

import bluetooth
import random
import struct
import time
import micropython

from ble_advertising import decode_services, decode_name

from micropython import const

_IRQ_CENTRAL_CONNECT = const(1)
_IRQ_CENTRAL_DISCONNECT = const(2)
_IRQ_GATTS_WRITE = const(3)
_IRQ_GATTS_READ_REQUEST = const(4)
_IRQ_SCAN_RESULT = const(5)
_IRQ_SCAN_DONE = const(6)
_IRQ_PERIPHERAL_CONNECT = const(7)
_IRQ_PERIPHERAL_DISCONNECT = const(8)
_IRQ_GATTC_SERVICE_RESULT = const(9)
_IRQ_GATTC_SERVICE_DONE = const(10)
_IRQ_GATTC_CHARACTERISTIC_RESULT = const(11)
_IRQ_GATTC_CHARACTERISTIC_DONE = const(12)
_IRQ_GATTC_DESCRIPTOR_RESULT = const(13)
_IRQ_GATTC_DESCRIPTOR_DONE = const(14)
_IRQ_GATTC_READ_RESULT = const(15)
_IRQ_GATTC_READ_DONE = const(16)
_IRQ_GATTC_WRITE_DONE = const(17)
_IRQ_GATTC_NOTIFY = const(18)
_IRQ_GATTC_INDICATE = const(19)

_ADV_IND = const(0x00)
_ADV_DIRECT_IND = const(0x01)
_ADV_SCAN_IND = const(0x02)
_ADV_NONCONN_IND = const(0x03)

_UART_SERVICE_UUID = bluetooth.UUID("6E400001-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_RX_CHAR_UUID = bluetooth.UUID("6E400002-B5A3-F393-E0A9-E50E24DCCA9E")
_UART_TX_CHAR_UUID = bluetooth.UUID("6E400003-B5A3-F393-E0A9-E50E24DCCA9E")

class BLESimpleCentral:
    def __init__(self, ble):
        self._ble = ble
        self._ble.active(True)
        self._ble.irq(self._irq)

        self._reset()

    def _reset(self):
[... 171 lines cut ...]
    print("Connected")

    def on_rx(v):
        print("RX", byte(v))

    central.on_notify(on_rx)

    with_response = False

    i = 0
    while central.is_connected():
        try:
            v = str(i) + "_"
            #print("TX", v)
            #central.write(v, with_response)
        except:
            print("TX failed")
        i += 1
        time.sleep_ms(400 if with_response else 30)

    print("Disconnected")

if __name__ == "__main__":
    demo()
`