micropython/micropython #6571: Issue when using __getattr__ with tuple.index (TypeError: argument has wrong type)

solarjoe, 2020-10-25:
I would like to use a simple workaround to have something like an Enum in Micropython:

```
class Enum(tuple): 
    __getattr__ = tuple.index
```
It is supposed to work like

```
State = Enum(['NO_MOVEMENT',
              'MOVEMENT'])

print(State.NO_MOVEMENT)
>>> 0
print(State[1])
>>> 'MOVEMENT'
```

It works in "normal" CPython, tested on 3.7, but in Micropython it throws

`TypeError: argument has wrong type`

If I write it like this

```
class Enum(tuple): 
    def __getattr__(self, y):
        return self.index(y)
```
it works fine. But I have no clue why, because this is basically the same code, right?

peterhinch, 2020-10-25:
This question would be better raised in [the forum](http://forum.micropython.org/index.php).

[this doc](https://github.com/micropython/micropython/wiki/Differences) suggests the probable answer:

> Subclassing of builtin types is partially implemented, but there may be various differences and compatibility issues with CPython. #401

solarjoe, 2020-10-25:
Ok, thanks.

solarjoe, 2020-10-25:
Reopened, I read https://github.com/micropython/micropython/issues/401 and at the end they mention that issues should be reported as as separate tickets.

jonnor, 2024-09-13:
There seems to be a pretty functional workaround for this problem. So perhaps we can close this?

closed 2025-03-03 by jonnor