micropython/micropython #6572: NRF52840: ADC is 8 bits, how to set to 12 bits?

JPFrancoia, 2020-10-25:
Hi,

I have been playing with the latest version of micropython on the nrf52840 USB dongle. The temperature over bluetooth example works perfectly for me (as a proof that micropython is properly flashed  on the board).

I wanted to try and read some values from the ADC pins. I wired the pin 0.29 to a potentiometer, and initialized an ADC object:

```python
from machine import ADC
adc = ADC(5)
adc.value()
```

However, I could only get values between 0 and 255 (0 when the potentiometer was turned to the minimum, 255 when turned to the maximum). This obviously means the ADC has an 8-bits resolution

The specs of the board tell us we can use  8, 10, 12 and even 14 bits of resolution (with oversampling): https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.0.pdf

I wasn't able to find a method in the ADC class to set the resolution of the ADC. Is it possible in the current state of micropython for the nrf boards?

zeekzhen, 2020-11-14:
It seems that the author is on vacation╭(●｀∀´●)╯

glennrub, 2020-11-15:
Hi @JPFrancoia, sorry for late answer.

However, you are right, it is 8-bit hard-coded in the ADC machine module. 
https://github.com/micropython/micropython/blob/master/ports/nrf/modules/machine/adc.c#L167

Should be quite easy to change this to 12-bit in the source for your current need. 

For a common implementation i could suggest defaulting to 10-bit which is common between nrf51(8, 9 and 10 bit), nrf52832 (8, 10, 12 and 14-bit), nrf52840 (8, 10, 12 and 14-bit) and nrf9160 (8, 10, 12 and 14-bit). For the future it could be more aligned with the [`pyb.ADCAll`](https://docs.micropython.org/en/latest/library/pyb.ADC.html) to provide a mechanism to set the resolution runtime. I'm not sure where to put this, `board` as a new class or in the `machine.ADC` class.

 @dpgeorge, would it be an option to add this parameter to the `machine.ADC` constructor?