micropython/micropython #6573: ESP32: Check the ESP-IDF error code

IhorNehrutsa, 2020-10-27:
**1) Is it correct to represent the ESP-IDF error code as OSError code?**
For example, for the code
```
from esp32 import RMT  
from machine import Pin  
rmt = RMT(1000, pin=Pin(10)) # 1000 is wrong  
```
we get an output
```
OSError: (-258, 'ESP_ERR_INVALID_ARG')
```
But I expect to see
```
EspError: (258, '0x102 ESP_ERR_INVALID_ARG')
```
(or something similar) according to errors 
```
#define ESP_ERR_INVALID_ARG         0x102   /*!< Invalid argument */
```
from the 
[esp_err.h#L31](https://github.com/espressif/esp-idf/blob/357a2776032299b8bc4044900a8f1d6950d7ce89/components/esp_common/include/esp_err.h#L31)

**2) Is this line of code correct? Do you need a '-' here?**
```
uint32_t pcode = -code;
```
from the
[mphalport.c#61](https://github.com/micropython/micropython/blob/05f95682e7ddfb08c317e83826df9a1d636676f3/ports/esp32/mphalport.c#L61)

**3) Is it possible to develop a C module with esp_err.py functionality?**
[esp_err.zip](https://github.com/micropython/micropython/files/5445514/esp_err.zip)

I try but it doesn't work completely.
[mod_esp_err.zip](https://github.com/micropython/micropython/files/5445585/mod_esp_err.zip)
I get TypeError when executing esp_err_test.py 
```
  File "<stdin>", line 9, in <module>
TypeError: can't create 'EspError' instances
```

jimmo, 2020-10-29:
> 1) Is it correct to represent the ESP-IDF error code as OSError code?

This is the way it's done, but whether it's the best thing to do... not sure.

> 2) Is this line of code correct? Do you need a '-' here?

The idea here is that positive values indicate a "real" error code (i.e. MP_Exxx), and negative values are custom error code (i.e. ESP-IDF codes).

> 3) Is it possible to develop a C module with esp_err.py functionality?

Your `mod_esp_err_EspError_type` instance doesn't have a `.make_new` entry and therefore cannot be constructed from Python. Here's roughly what that would look like.

```c
STATIC mp_obj_t mod_esp_err_EspError_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *all_args) {
    ...allocate and return an instance of _mp_obj_EspError_t
}

STATIC const mp_obj_type_t mod_esp_err_EspError_type = {
    { &mp_type_type },
    .name = MP_QSTR_EspError,
    .locals_dict = (void*)&mod_esp_err_EspError_locals_dict,
    .make_new = mod_esp_err_EspError_make_new,
};
```

IhorNehrutsa, 2020-10-30:
Thanks, Jim.
I have developed an external C-module for ESP errors.
[mod_esp_err.zip](https://github.com/micropython/micropython/files/5463664/mod_esp_err.zip)
When executed
[esp_err_test.zip](https://github.com/micropython/micropython/files/5463736/esp_err_test.zip)
EspError raised as expected
```
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
EspError: (258, '0x102 - ESP_ERR_INVALID_ARG')
```
WIP

IhorNehrutsa, 2020-11-16:
This issue solved in PR #6626

dpgeorge, 2021-04-14:
I'll close this issue because points (2) and (3) are answered above.  And point (1) is covered by #6638.

closed 2021-04-14 by dpgeorge

QGB, 2023-04-23:
idf.py fullclean

make BOARD=GENERIC_C3